OllisGit/OctoPrint-DisplayLayerProgress #168: rename plugin in pip list

RogierSchuring, 2020-08-26:
Hi OllisGit,

I do not know if it's possible for you and how much work it would be, but could you rename the plugin so that it will have the same naming convention as the other octoprint plugins while doing a pip list?

Currently:
python -m pip list|grep Disp
DisplayLayerProgress          1.23.0

Others:
python -m pip list|grep OctoPrint
OctoPrint                     1.4.2
OctoPrint-Dashboard           1.14.0
OctoPrint-ExcludeRegionPlugin 0.2.0
OctoPrint-FileCheck           2020.8.7
OctoPrint-FirmwareCheck       2020.6.22
OctoPrint-OpiTemp             1.0.8
OctoPrint-PrintTimeGenius     2.2.5
OctoPrint-Telegram            1.5.2

This way it's easier to automate octoprint deployment and plugin installation.

Thanks!

OllisGit, 2020-08-26:
Hi @RogierSchuring ,

the result of my pip list looks like this and it includes a lot of other plugins not using such naming-pattern.
[image]

I will not do it, because I don't see the benefit and also the risk is high that I broke thousand of current installations, sorry!

closed 2020-08-26 by OllisGit

blueflame2020, 2020-08-28:
Okay, I understand. I will work around this in my automation :)

RogierSchuring, 2020-08-28:
Okay, I understand. I will work around this in my automation then :)