OllisGit/OctoPrint-DisplayLayerProgress #169: New update breaks insertion of layers function

Crowlord, 2020-08-30:
1.23.0 update

On update the layer insertion feature triggers the following error in Prusa Slic3r.

Error uploading to print host:
HTTP 500: <!DOCTYPE HTML PUBLIC "-//W3C//DTD HTML 3.2 Final//EN">
<title>500 Internal Server Error</title>
<h1>Internal Server Error</h1>
<p>The server encountered an internal error and was unable to complete your request. Either the server is overloaded or there is an error in the application.</p>

Was fine before update and switching of feature allows upload as before.

ifekali, 2020-08-30:
Same here. Unfortunately this breaks the Dashboard plugin also, depending on this one.

Until fixed, is there a way to revert to previous version?

Crowlord, 2020-08-30:
Also occurs if directly uploaded 

[image]

OllisGit, 2020-08-30:
SORRY!!!

New version 1.23.2 is out and fix the upload functionality!

Crowlord, 2020-08-30:
Confirmed fixed

closed 2020-08-30 by Crowlord