OllisGit/OctoPrint-DisplayLayerProgress #170: Can't upload files anymore (Exception)

markusressel, 2020-08-30:
When trying to upload a file to Octoprint the action fails with an internal server error:

```
Aug 30 11:33:14 3DPrinter octoprint[37623]: 2020-08-30 11:33:14,373 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. Checking LayerExpressions.
Aug 30 11:33:14 3DPrinter octoprint[37623]: 2020-08-30 11:33:14,373 - octoprint.plugins.DisplayLayerProgress - INFO - FilePreProcessor. LayerExpression valid. Start processing...
Aug 30 11:33:14 3DPrinter octoprint[37623]: 2020-08-30 11:33:14,378 - octoprint - ERROR - Exception on /api/files/local [POST]
Aug 30 11:33:14 3DPrinter octoprint[37623]: Traceback (most recent call last):
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/flask/app.py", line 2447, in wsgi_app
Aug 30 11:33:14 3DPrinter octoprint[37623]:     response = self.full_dispatch_request()
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/flask/app.py", line 1952, in full_dispatch_request
Aug 30 11:33:14 3DPrinter octoprint[37623]:     rv = self.handle_user_exception(e)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/flask/app.py", line 1821, in handle_user_exception
Aug 30 11:33:14 3DPrinter octoprint[37623]:     reraise(exc_type, exc_value, tb)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/flask/_compat.py", line 39, in reraise
Aug 30 11:33:14 3DPrinter octoprint[37623]:     raise value
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/flask/app.py", line 1950, in full_dispatch_request
Aug 30 11:33:14 3DPrinter octoprint[37623]:     rv = self.dispatch_request()
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/flask/app.py", line 1936, in dispatch_request
Aug 30 11:33:14 3DPrinter octoprint[37623]:     return self.view_functions[rule.endpoint](**req.view_args)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint/server/util/flask.py", line 1315, in decorated_view
Aug 30 11:33:14 3DPrinter octoprint[37623]:     return func(*args, **kwargs)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint/vendor/flask_principal.py", line 199, in _decorated
Aug 30 11:33:14 3DPrinter octoprint[37623]:     rv = f(*args, **kw)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint/server/api/files.py", line 409, in uploadGcodeFile
Aug 30 11:33:14 3DPrinter octoprint[37623]:     added_file = fileManager.add_file(FileDestinations.LOCAL, futureFullPathInStorage, upload,
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint/filemanager/__init__.py", line 501, in add_file
Aug 30 11:33:14 3DPrinter octoprint[37623]:     path_in_storage = self._storage(destination).add_file(path, file_object, links=links, printer_profile=printer_profile, allow_overwrite=allow_overwri>
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint/filemanager/storage.py", line 774, in add_file
Aug 30 11:33:14 3DPrinter octoprint[37623]:     file_object.save(file_path)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint/filemanager/util.py", line 92, in save
Aug 30 11:33:14 3DPrinter octoprint[37623]:     shutil.copyfileobj(source, dest)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/usr/lib/python3.8/shutil.py", line 202, in copyfileobj
Aug 30 11:33:14 3DPrinter octoprint[37623]:     buf = fsrc_read(length)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint/filemanager/util.py", line 198, in read
Aug 30 11:33:14 3DPrinter octoprint[37623]:     processed_line = self.process_line(line)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 166, in process_line
Aug 30 11:33:14 3DPrinter octoprint[37623]:     line = self._modifyLineIfLayerComment(inputLine, layerExpression)
Aug 30 11:33:14 3DPrinter octoprint[37623]:   File "/home/markus/venvs/octoprint/lib/python3.8/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 194, in _modifyLineIfLayerComment
Aug 30 11:33:14 3DPrinter octoprint[37623]:     currentLayer = str(self.currentLayerCount)
Aug 30 11:33:14 3DPrinter octoprint[37623]: AttributeError: 'LayerDetectorFileProcessor' object has no attribute 'currentLayerCount'
Aug 30 11:33:14 3DPrinter octoprint[37623]: 2020-08-30 11:33:14,384 - tornado.access - ERROR - 500 POST /api/files/local (192.168.2.91) 193.17ms
```

stony2k, 2020-08-30:
Same issue here. After disabling the plugin, the fileupload works again. Do you need my logfiles or do markusressel's logfile enough?

energiebrand, 2020-08-30:
I had opened issue #166 and did several tests on 2 OctoPi installations without any problem with the new release.
I´m using Python 2.7.13. What Python version do you use? Maybe there is the problem...?
You can check with Putty and command " python --version" .

[image]

kats-greg, 2020-08-30:
[image]

energiebrand, 2020-08-30:
OK, have seen that markusressel did use Python 3.8 and thought that could be a probelm. Let´s wait for Olli...

OllisGit, 2020-08-30:
SORRY!!!

New version 1.23.2 is out and fix the upload functionality!

kats-greg, 2020-08-30:
OK.  I was able to get the latest update installed, and upload a file without issue.

Thanks

markusressel, 2020-08-31:
Can confirm, on new version 1.23.2 the file upload works again.

closed 2020-08-31 by markusressel