OllisGit/OctoPrint-DisplayLayerProgress #172: Print button clickable while print is running

zaphod82, 2020-09-26:
After installing plugin, the print button remains clickable when printer is printing.  Removing the plugin resolves the issue.  Adding the plugin back and issue reappears.

zaphod82, 2020-09-26:
[image]

Screenshot of issue.

zaphod82, 2020-09-26:
DisplayLayerProgress Plugin (1.23.3)
OctoPrint version : 1.4.2
OctoPi version    : 0.17.0

OllisGit, 2020-09-27:
Hi @zaphod82,

thx for reporting the issue. 
The fix will be in the next release. Because it is not a critical bug it could take some time when I release a new version with more fixes or features.

zaphod82, 2020-10-01:
Understandable.  Just wanted to report it as if things aren't reported, no one knows about them and they will never be fixed.

OllisGit, 2020-10-07:
Hi @zaphod82 ,
it is now fixed in Version 1.24.0.

Please test and give feedback.

Thx, in advance
Olli

closed 2020-11-16 by stale[bot]