OllisGit/OctoPrint-DisplayLayerProgress #173: Stuttering

Matejz90, 2020-10-02:
Hello there!

First sorry for my English because is not my native language and i don't use everyday.

On ender 3 pro (with mks sgen and skr 1.4 but i think this doesent matter, also if is ender) i have terrible stuttering. With clean install (raspbian + octoprint + klipper) i have in CPU usage between 10-30% for octoprint process. It is on raspby 3b+. 
Most time is usage somewhere at middle. When i install DLP then i suddenly get CPU usage to 120% (how is this possible i dont know, but only when printing). When process hits 120+ then printer completly freeze for few miliseconds or sometimes seconds too till usage drop to around 30%. This can happen in 1 minute 1 time or 5 times and is completly random. I use top command in shell. Raspby don't show termal issues or something. 

Have anyone solution for capping DLP process to i don't know 50 or 70% usage?
I tryed also with 1.19.0 version and it is same. I don't want go digg too much in code because i know that i will be 1 month only in this. 

Thanks anyone and thanks creator for this amazing plugin!

OllisGit, 2020-10-02:
Hi @Matejz90 ,

please try to use a far older Version 1.6.0. https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.16.0/master.zip

gdombiak, 2020-10-08:
Hey @Matejz90 ,

Could you try the latest release that includes some important optimizations that will reduce cpu impact of this plugin? Prints with lots of curves will benefit greatly from this version and reduce stuttering as well. Let us know of the new version helped or even solved this issue. 

Thanks 
Gaston

Matejz90, 2020-10-08:
Now i don't have something complex for printing but some toys for my daughter and looks like it is stable now with latest version...

[image]

gdombiak, 2020-10-08:
That's great to hear @Matejz90! Let us know how it goes with a complex print.

Thanks,
Gaston

closed 2020-11-17 by stale[bot]