OllisGit/OctoPrint-DisplayLayerProgress #174: Error when printing from SD card.

Mrnt, 2020-10-02:
Not sure if this is by design, but the following steps:

1. Use "Upload to SD" button to upload file to printer's SD card
2. Issue command `M23 <some_path_to_file_on_SD_card>`

Cause the following error in octoprint.log:
```
2020-10-02 15:50:13,568 - octoprint.plugin - ERROR - Error while calling plugin DisplayLayerProgress
Traceback (most recent call last):
  File "/Users/gavin/Development/OctoPrint/src/octoprint/plugin/__init__.py", line 230, in call_plugin
    result = getattr(plugin, method)(*args, **kwargs)
  File "/Users/gavin/Development/OctoPrint/venv3/lib/python3.8/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 1129, in on_event
    selectedFile = self._file_manager.path_on_disk(fileLocation, selectedFilename)
  File "/Users/gavin/Development/OctoPrint/src/octoprint/filemanager/__init__.py", line 688, in path_on_disk
    return self._storage(destination).path_on_disk(path)
  File "/Users/gavin/Development/OctoPrint/src/octoprint/filemanager/__init__.py", line 716, in _storage
    raise NoSuchStorage("No storage configured for destination {destination}".format(**locals()))
octoprint.filemanager.NoSuchStorage: No storage configured for destination sdcard
```

OllisGit, 2020-10-07:
Hi @Mrnt,
what system do you use, a RaspberryPi?

I will fix this error-message in the next release, but please keep in mind that DLP did not support files located on SD-Card.

I raised a question in the form, maybe there are some ideas how to read the sd-card file content:
https://community.octoprint.org/t/read-sd-card-gcode-content-during-file-selection/24933

OllisGit, 2020-10-07:
The new version 1.24.0 could now handle SD-Card prints.

Please test and give feedback.

Thx, in advance
Olli

Mrnt, 2020-10-09:
Hi Olli, that did the trick. Thanks!

closed 2020-10-09 by Mrnt