OllisGit/OctoPrint-DisplayLayerProgress #176: Printer Display info pop-up always shows even if "Printer Display" is unchecked

watermelonpizza, 2020-10-08:
This seems to be a similar/same error to #66? 
I have installed the plugin version 1.24.0 on octoprint 1.4.2.

If I untick the show setting I just get an empty pop-up.
[image]

jonathan-schmieg, 2020-11-18:
this bug shouldn't have been closed - it still exists

watermelonpizza, 2020-11-18:
Yeah, :( there has been no progress or anything, I guess?
I can't reopen it.

OllisGit, 2021-01-13:
Hi!

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

After removing the checkbox ```Show printer display on Desktop``` did you close the popup with the X in the top right corner?

Then it should not appear again.

I will try to fix this in one of the next releases. Toggling the checkbox should directly reflected to the visibillity of the popup.

OllisGit, 2021-01-30:
Hi @watermelonpizza,

new release 1.25.0 is out.
Please test and give feedback.

Thx, in advance
Olli

jonathan-schmieg, 2021-02-04:
@OllisGit I can confirm that the issue is solved. Thank you very much!

closed 2021-02-07 by OllisGit