OllisGit/OctoPrint-DisplayLayerProgress #177: Only show in browser tab title when printing

TomW1605, 2020-10-17:
at the moment the browser tab title is overridden and shows `0% -` when there is no print running. this just makes it look like an error. it would be great if it only displayed this when there is actually some info to show.

also an option to make this time remaining not ETA would be great.

TomW1605, 2020-11-19:
still waiting for a response. not trying to rush you just keeping it open

OllisGit, 2021-01-13:
Hi @TomW1605 !
PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

I am not sure why `0% -` looks like an error?!? Please explain.

What do you mean with your second point? You can change the default tab output to:
```
[progress]% [printtime_left]
```
Is that what you want?

TomW1605, 2021-01-14:
hi
i also took a break in printing and in that time this feature was added so when i came back and updated everything i thought it was an error. it does still look like it is missing a placeholder with the `-` that has nothing after it. it is also unnecessary to show that information when there is nothing printing. it would be better to make it something like `Octoprint - Not Printing` as this gives more actual information about the current status.

also yes that is exactly what i was looking for with the second point.

OllisGit, 2021-01-29:
hmmm... I could add something like this to the tab:
```
[progress]% [estimated_end_time] ([printer_status])
```
and the result looks like this:
```
0% - (idle)
```
Should be easy to implement.

TomW1605, 2021-01-30:
that would look a lot better. although i do think you might as well just release the title when not printing and let it default back to the standard one.

OllisGit, 2021-01-30:
Hi @ssendev,

new release 1.25.0 is out.
Now the print-state is visible in the tab.
Please test and give feedback.

Thx, in advance
Olli

TomW1605, 2021-02-01:
that works however when nothing is printed the state is operational not idle. this again suggests an error as operational implies (at least to me) that the printer should be doing something.
also for some reason this placeholder string `[progress]% [estimated_end_time] ([printer_state])` gives this title `0% - (operational)`. where did the `-` come from?

OllisGit, 2021-02-02:
Hi @TomW1605,

the output "operations" is directly reflected from OP, see https://docs.octoprint.org/en/master/modules/printer.html?highlight=operational#octoprint.printer.PrinterInterface.get_state_id

I filtered/mapped some states, like this:
```
                if (currentPrinterStateId == "PRINTING"):
                    printerState = "printing"
                elif (currentPrinterStateId == "OPERATIONAL"):
                    printerState = "operational"
                elif (currentPrinterStateId == "OFFLINE"):
                    printerState = "offline"
                elif (currentPrinterStateId == "PAUSED"):
                    printerState = "paused"
                elif (currentPrinterStateId == "ERROR" or currentPrinterStateId == "CLOSED_WITH_ERROR"):
                    printerState = "error"
                else:
                    printerState = "idle"
```

- is the default output if a value is not present.

TomW1605, 2021-02-03:
ok that makes sense. i think that is as good as its going to be. thanks for making this change.

closed 2021-02-03 by TomW1605