OllisGit/OctoPrint-DisplayLayerProgress #178: DLP always shows last print progress

mordhau5, 2020-10-20:
One thing I've noticed with DLP across several of my printers is that once a print completes, the Browser Tab and Nav Bar always show the last message they sent (which is usually something like "100% HH:MM). Is there any way to have it clear after a couple minutes, or manually even?

OllisGit, 2021-01-13:
Hi @mordhau5 ,

why is this needed? I don't see a benefit.

PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

mordhau5, 2021-01-14:
@OllisGit When a print is cancelled the header will read the percentage at which the print was stopped until a new one is started or the server restarted. Or when a print is completed, there's no reason for it to read 100% when I come back the next day. Why should the tab keep that information in the tab title? The purpose of having info in the tab title is to allow quick-checks on progress at a glance and if I print has failed, is cancelled or completes, that info isn't needed anymore.

OllisGit, 2021-01-29:
Hi @mordhau5,

did you already tested the gcode script-feture from Octoprint?
[image]

closed 2021-03-12 by stale[bot]