OllisGit/OctoPrint-DisplayLayerProgress #180: Ender 3 V2 compatibility?

DarkZed1, 2020-10-23:
Is there a way to make this compatible with the Ender 3 V2 display?

closed 2020-12-02 by stale[bot]

OllisGit, 2021-01-13:
Hi @DarkZed1 ,

could you send M117 messages to that display? if yes, it should work.
But there are printers that could not do that, see https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/18

agarthand, 2021-06-11:
I had just fall into the same situation. Received my new Ender 3 v2 and was wondering that no progress is shown (was used to it from my prior ANet A8). Thereof I just tried to send the manual "M117 99% 12 of 210" to the printer but it mades no update to the display. At least not with the (uptodate) Creality Vendors firmware (1.0.2) Maybe this requires a customized Marlin FW?