OllisGit/OctoPrint-DisplayLayerProgress #181: Max Height not calculated when files sent directly from Slic3r

jackmang, 2020-10-24:
Related to #166 
I'm using the latest release DisplayLayerProgress Plugin (1.24.0)
When I print in OP, I typically send the files directly from Slic3r which works great.  When I do this with your plug-in everything works except for total height.  I find that it displays the total height as "-".
I reviewed bug #166 and found the mention of dragging the gcode from explorer and dropping into OP.  I tried that and it works properly.
Directly sent from Slic3r:
[image]

Drag and dropped same Slic3r file from Explorer to OP:
[image]

In both cases if I download the resulting gcode from OP I get an identical file.
gcode attached.
[test.gcode.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5433845/test.gcode.txt)

jackmang, 2020-10-25:
I just did additional testing and I'm not getting a consistent result.  Before I waste your time let me do some additional confirmations to see if I can find a pattern.  I'll close this and resubmit if and when I have more to report.

closed 2020-10-25 by jackmang

moogster, 2021-10-20:
Did you ever sort this issue out?

OllisGit, 2021-10-20:
fyi: the max height value is provided by octoprint and the value is normaly calculated in the background (if not deactivated "never"), see plugin-settings:
[image]
Sometimes..when you use bigger models...the value is not shown, because it take some time. So you need to wait select an other file and reselect the file again.