OllisGit/OctoPrint-DisplayLayerProgress #182: Show current system time in NavBar

charredchar, 2020-10-31:
I have a monitor at my workbench that is using my Pi's browser in Kiosk Mode for OctroPrint, because of this I do not have any information on the screen that isn't on the OctoPrint dashboard, this includes the current time and date. I would really love to be able to add this to the top navigation bar using DisplayLayerProgress as I don't need yet another place that shows ETAs, progress, etc.

I also wanted to mention, I love having the progress in the Browser Title, thanks!!

OllisGit, 2021-01-13:
Hi @charredchar,

nice idea! 
Should be easy to implement (maybe some challenges with timezones, so I added this feature to my short-term list.

charredchar, 2021-01-13:
> Hi @charredchar,
> 
> nice idea!
> Should be easy to implement (maybe some challenges with timezones), so I added this feature to my short-term list.

If timezones could be an issue would it be easier to pull the time from the currently viewing system or even the system running OctoPrint? That way you wouldn't need to configure it within the plugin and let the underlaying system do all the work in regards to timezones and daylight savings and whatnot.

OllisGit, 2021-01-30:
Hi @charredchar ,

new release 1.25.0 is out. 
Now you can turn on a clock in the navbar (default: off) 
Please test and give feedback.

Thx, in advance
Olli

charredchar, 2021-02-13:
Pretty neat, thanks!

I did come across one issue though, during a reload of the page the time disappears. To get it to show up I need to go into the plugin settings and change a time display settings such as changing from right to left.

OllisGit, 2021-04-04:
Hi @charredchar,

fyi: display issue will be fixed in next release.

OllisGit, 2021-04-05:
Hi @charredchar,

new release 1.26.0 is out, please test and give feedback.

Thx, in advance
Olli

charredchar, 2021-05-03:
Sorry for it taking so long to reply, but I've had no issue with it disappearing since the update, thanks so much as it is so useful for me!

The only issue left is it seems to ignore the left or right side options, though I have not tested yet if this is caused by one of my theme plugins or not. I was also wondering if it would be possible for me to change it to a 24H clock instead of a 12H? Is it based off anything else that I would need to change for it to switch?

OllisGit, 2021-05-10:
Hi @charredchar,

I added an option to format the clock output (will be in the next release).
Left/Right: I used a html-element with the id: dlpNavBarTime-left and right to place the clock. Maybe one of your plugin changed/removed such element.

OllisGit, 2021-10-14:
Hi @schmjop,

the new release 1.27.0 with this feature is out now. See also https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/224
Please test and give feedback.

Thx, in advance
Olli

closed 2021-11-24 by stale[bot]