OllisGit/OctoPrint-DisplayLayerProgress #185: Display Lyaer plugin is trgerring a filament change

LordDarkneo, 2020-11-26:
Hello there,

Since the last update, I think I figoured out an issue with the version 1.24.0. When starting a print, hereunder what's triggered by the plugin:

`2020-11-26 08:42:37,111 - octoprint.events.fire - DEBUG - Firing event: DisplayLayerProgress_feedrateChanged (Payload: {'changeFilamentTimeLeft': '-', 'feedrate': '9000', 'updateReason': 'feedrateChanged', 'totalHeight': '14.6', 'totalLayer': '73', 'estimatedEndTime': '21:37', 'changeFilamentCount': 0, 'estimatedChangedFilamentTime': '-', 'printTimeLeftInSeconds': 46478, 'totalHeightFormatted': '14.6', 'currentHeight': '1.80', 'printTimeLeft': u'12h54m38s', 'lastLayerDuration': '0h:17m:36s', 'averageLayerDurationInSeconds': 864, 'lastLayerDurationInSeconds': 1056, 'changeFilamentTimeLeftInSeconds': 0, 'averageLayerDuration': '0h:14m:24s', 'feedrateG1': '1500', 'feedrateG0': '9000', 'fanspeed': '100%', 'progress': '16', 'currentHeightFormatted': '1.8', 'currentLayer': '8'})
`
thus I have a M600 like trigerred on the printer (and not on octoprint). So i absolutely need to be close to the printer (because the filament is pulled out).

Could you please tell me where and how to fix this? 

[octoprint (4).log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5604047/octoprint.4.log)

OllisGit, 2021-01-13:
Hi @LordDarkneo ,

I am not sure why you think DLP is responsible for filament change.

The mentioned log-message only shows that the plugins sends an Event called ```DisplayLayerProgress_feedrateChanged```to the eventbus of OctoPrint. 
Other plugins (e.g. Dashboard) could isten on this event and there own thing.

Please explain a little bit deeper the issue with DLP.
PS. Sorry, for being late to the party, but I made a "Octoprint sabatical" (no coding, no printing) the last months.

LordDarkneo, 2021-02-22:
Hi everyone,
Sorry for late answer on this topic. I performed many updates/upgrade meanwhile, and the issue has gone... I was thinking about DLP because in the log files, nothing else was triggered for a filament change... 
I'll close this issue

closed 2021-02-22 by LordDarkneo