OllisGit/OctoPrint-DisplayLayerProgress #186: Layer Indicator not found even with sucessful custom regex

Delarado, 2020-12-01:
Hi,

When I upload my Gcode I get:

```
Attention
Layer indicator not found in file: 'Mini_GhostBody_TPU.gcode'
Check layer pattern in DisplayLayerProgress-Settings and reupload the file!
```

However, I have the following configured:

`1		[; move to next layer \(([0-9]+)\).*]		Test`

Sample Gcode:

`G1 E22.60367 F2400.00000 ; retract
G92 E0 ; reset extrusion distance
G1 Z0.750 F3600.000 ; move to next layer (2)
G1 X45.733 Y59.266 ; move to first perimeter point
G1 E5.00000 F2400.00000 ; unretract`

As far as I can see, my regular expression is matching, but the gcode remains unchanged upon upload?

Complete Gcode is attached, as re-downloaded after upload to the octoprint server.
[post_Mini_GhostBody_TPU.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5624793/post_Mini_GhostBody_TPU.gcode.zip)

Delarado, 2020-12-01:
Also note I tried the regular expression without the .* at the end, and the sample gcode snippet is not showing the new lines (The GitHub software seems to have deleted them)

closed 2021-01-10 by stale[bot]

OllisGit, 2021-01-13:
Hi @Delarado ,

for performance reason the scanned line must start with a comment symbol (;). In your example the semicolon is in the middle of the line.

I updated my wiki: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works