OllisGit/OctoPrint-DisplayLayerProgress #187: Problem with Total Layer Height

CBoismenu, 2020-12-20:
Hi 

I seem to be having a problem with the Total Height display of the plugin.
For exemple if I print an object with a 0.2mm layer height and the object has 150 layers the total layer height should be 30.0 mm
When I start printing or lod a file the plugin displays the proper # of layers (150) but displays 1 layer less then expected (29.8 mm) for the total height.
I wouldn't be suprise if it is something thta I didn't set up properly.
Would appreciate if you could point me in the right direction.

Here are the version of what I'm using:

OctoPrint 1.5.2
DisplayLayerProgress Plugin 1.24.0
PrusaSlicer 2.2.0

Thanks for your help,

OllisGit, 2021-01-13:
Hi @CBoismenu ,

please provide your gcode, so that I can reproduce the issue.

Thx,
Olli

CBoismenu, 2021-01-13:
> <OctoPrint-DisplayLayerProgress@noreply.github.com>
> <mention@noreply.github.com>
> Layer Height (#187)

Hi,

 

I have attached a gcode of a 25 mm cube.

 

Whatever I slice always does exactly the same thing, the total number of
layers is spot on but the total height is off by one layer.

 

In this case the total # of layers displayed in OctoPrint is 125 Layers bt
the Total Height is 24.8 mm (the actual item is 25 x 25 x25).

 

Im almost certain that I may have a self inflicted setup problem.

 

Thanks for your help,

<https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/187#issue
XIILANCNFSM4VDKWAXA> .
<https://github.com/notifications/beacon/AJJYG4KTIGTKLM4LFV67JBLSZXIILA5CNFS
M4VDKWAXKYY3PNVWWK3TUL52HS4DFVREXG43VMVBW63LNMVXHJKTDN5WW2ZLOORPWSZGOFVDHR2A
.gif>

OllisGit, 2021-01-29:
Hi @CBoismenu,

you can't use the reply-function from your mail-client to upload an attachment to an issue. Do this in the github ui!
Btw. EMail-Content in an issue looks really ugly, don't do this.
Btw2. Now everybody can unsubscribe your account from notifications 

Thx, in advance
Olli

OllisGit, 2021-01-29:
I was able to repair the zip.

For the dice I received a value of 19.8mm from OctoPrint:
[image]

The plugin is not responsible for the height it is provided by OctoPrint.

Jack77777777, 2021-01-31:
I have the same issue since i installed this Plugin. Using Octoprint 1.5.3. and Superslicer (fork of Prusaslicer). Increasing Layer offset shows more layers but total height is always -0.2mm.

OllisGit, 2021-02-02:
Hi @Jack77777777,

please send me a sample gcode with:
- slicer height
- OP gcode viewer model-info height
- DLP height
...and which value is the truth?

Jack77777777, 2021-02-03:
Hi @OllisGit ,

Thank you for the reply. Attached are screenshots and the gcode. 
- Superslicer shows 75 Layers and 15mm height->that´s the correct one
- Gcode Viewer shows 76 Layers and 14.8mm height
- DLP shows 75 Layers and 14.8mm height
**Knob for nut_SS.gcode.txt** is gcode exported from Superslicer
**Knob for nut_SS.gcode.txt** is gcode downloaded from OP after sending with Superslicer
Just delete .txt

[Knob for nut_SS.gcode.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5917366/Knob.for.nut_SS.gcode.txt)
[Knob for nut_OP.gcode.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5917368/Knob.for.nut_OP.gcode.txt)
[image]
[image]

Jack77777777, 2021-02-05:
I created a very simple test model with 1 Layer height (0.2mm). DLP shows 0.1mm Total height.
Screenshots and Gcodes attached (remove .txt)
[image]
[image]
[Test_TPLink v1_OP.gcode.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5931920/Test_TPLink.v1_OP.gcode.txt)
[Test_TPLink v1_SS.gcode.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5931921/Test_TPLink.v1_SS.gcode.txt)

OllisGit, 2021-02-10:
Hi @Jack77777777,

from your provided informations I can see that the height in the sidebar (provided by DLP) is equal to Model-Height (provided by OP), only the layer count is different.

The model layer number is starting at "one" and evaluated during gcode interpretation. The sidebar layernumber is started at zero (could be adjusted in plugin-settings) and parsed from gcode layer comments.

Jack77777777, 2021-02-10:
Hi @OllisGit Which plugin do you mean. In the past I have already set Add layer offset to 1 in DLP settings with the result that total height is shwoing 1 Layer less.

OllisGit, 2021-02-13:
Hi @Jack77777777,

layer offset settings has nothing to do with hight.
[image]
If you increase the layer offset, the layer-number in the sidebar will increase. if not then it is a bug.

Because the DLP-Plugin reads the height value from the metadata (location: ```/uploads/.metadata.json```) provided by OctoPrint, you need to raise a ticket at OctoPrint about the missing 0.2mm.

Please open the metafile, look for your gcode and compare the values.

Jack77777777, 2021-02-14:
I can't find the height in gcode. Any tipp what to look/search for ?

Jack77777777, 2021-03-01:
https://github.com/OctoPrint/OctoPrint/issues/4036

Jack77777777, 2021-03-04:
@OllisGit will be fixed in OP 1.6.0

This can be closed then.

closed 2021-04-04 by OllisGit