OllisGit/OctoPrint-DisplayLayerProgress #188: Files loaded through Octoprint-Gitfiles

M1XZG, 2020-12-26:
I'm syncing files using my modified version of [OctoPrint-GitFiles](https://github.com/M1XZG/OctoPrint-GitFiles), so files are synced from GitHub to my octoprint. When I upload gcode direct to OctoPrint DLP will add the layer indictors fine, but when I sync via Git these aren't add, is there a way that I can force these files to be modified after a git sync so they will have the layer indicators?

OllisGit, 2021-01-13:
Hi @M1XZG ,

currently there is no way to do this. Like in the [description](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works) it only adds the layer indicator during upload (UI/API).

.... could be an enhancement....do we need a single button for each file in the Files-View or should all current files scanned at once...any ideas how the UI should act?

M1XZG, 2021-01-13:
Having thought about it a bit more, I'm not sure really how this would work with the gitfile plugin .. if you sync files from GitHub THEN apply the layer indication that then means the files are now out of sync with GitHub and the gitfiles plugin could well pull down from the repo again which would overwrite the changes by DLP.

OllisGit, 2021-01-30:
Hi @M1XZG,

I looked into the git-Plugin and it use a normal pull-command, so each file which is modified locally (on the OP-Server) will stay the same....except you modified the same file in the git repository, then git will merge the file (maybe with a merge conflict).

One solution could be that DLP modifies the gcode everytime before printing..could take some time....and the git plugin always force a pull und drop all local changes.

closed 2021-03-12 by stale[bot]