OllisGit/OctoPrint-DisplayLayerProgress #191: Filament Sensor, remaining

Alexwijn, 2021-01-06:
I got the BIGTREETECH Smart Filament Sensor installed on my Ender 3 V2 and I receive messages like this:
`Recv: Remaining mm: 6.90`. I was wondering if I can get the number 6.90 as progress information. Thanks!

OllisGit, 2021-01-13:
Hi @Alexwijn ,

sorry, but the plugin only reads the gcode that is sent to the printer. There is no listener for printer messages.

Do you know if other sensors also send the processed length to the terminal?

Alexwijn, 2021-02-12:
I created an own plugin that uses the `octoprint.comm.protocol.gcode.received` hook and send out a custom event. For me all events are sent to MQTT, so I can read it from my Home Assistant and notify me when something is about to go wrong.

closed 2021-02-12 by OllisGit