OllisGit/OctoPrint-DisplayLayerProgress #192: Enhancement request - Send M117 and beginning of layer and every n seconds after that

Sophist-UK, 2021-01-09:
I print a mixture of small and large objects. Small objects take only seconds per layer, large objects can take 10s of minutes per layer.

I would really like to see my printer screen updated with an M117 at the beginning of every layer regardless of how long since the last M117 was sent, and then have another M117 sent x minutes later if we haven't started a new layer since.

Is this possible please?

OllisGit, 2021-01-13:
Hi @Sophist-UK ,

have you tried the [Dashboard-Plugin](https://plugins.octoprint.org/plugins/dashboard/) ? There is a Layer-Duration graph.

Sophist-UK, 2021-01-13:
@OllisGit Thanks for the pointer - which I will definitely use, however that is not a fix for the message issue identified.

Sophist-UK, 2021-02-13:
I would still like this issue to be addressed.

OllisGit, 2021-02-28:
Hi @Sophist-UK,

fyi: I added your idea to my "Idea-Planing-Board".

Sophist-UK, 2021-03-30:
I still would like this implemented. Bumping to stop the bot from closing it.

closed 2021-11-08 by stale[bot]