OllisGit/OctoPrint-DisplayLayerProgress #193: [Request] Support of M73 progress

OllisGit, 2021-01-12:
See https://github.com/OllisGit/OctoPrint-CostEstimation/issues/7

There should be an option to show the M73 progress values provided from the sliced gcode.

Easiest implementation:
- Add new keyword beside ```[progress]```, just add ```[M73progress]```
- Listen for M73 GCode and use this value output-pattern

OllisGit, 2021-01-17:
Hi @WarrenSchultz,
can you please provide a sample g-code file with included M73 commands?

Thx, in advance
Olli

WarrenSchultz, 2021-01-18:
Sure, here go go:

[3DBenchy_Repaired_0.2mm_PLA_MK3SMMU2S_1h27m.gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5832086/3DBenchy_Repaired_0.2mm_PLA_MK3SMMU2S_1h27m.gcode.zip)

OllisGit, 2021-01-30:
Hi @WarrenSchultz ,

new release 1.25.0 is out.
Please test and give feedback.

Thx, in advance
Olli

WarrenSchultz, 2021-01-31:
Thanks for looking into this. The filament manager plugin is currently not deploying properly (bugs are logged there already), so I can't test it yet fully. 

The Dashboard plugin is reporting this error with the new version, not sure if that's a bug on this side or that one.
[image]

Actually, looking at this, it appears that this isn't deploying correctly either. It names the plugin Unknown when requesting a restart post-install attempt.

OllisGit, 2021-02-02:
Hi @WarrenSchultz,

is the DashBoard-Plugin working with lates DLP 1.25.3?

WarrenSchultz, 2021-02-03:
Looks like it, thanks!

closed 2021-02-07 by OllisGit

hayden-t, 2023-01-02:
Hi, I have 1.28 and there seems to be no m73 sent ?? i dont see any options mentioning it either ? I get the stats via m117 though.