OllisGit/OctoPrint-DisplayLayerProgress #196: Layer Error - Layer: - [blank] of - [blank]

spankurmonkey, 2021-01-25:
Every time I start a print I get a layer error.

Top of the UI says :  Layer Error - Layer: - [blank] of - [blank]
[image]

I use the S3D Slicer- apparently I don't have the correct post processing script for layer as it relates to Octoprint... this is what I have-
[image]

example from gcode
`G90
M82
M106 S0
M140 S50
M190 S50
M104 S190 T0
M109 S190 T0
M75 ; Start Print Timer and Engage Fil Sensor
M84 E ; Disable E Motor for probe accuracy on direct drive systems
G28 ; home all axes
G28 Z ; home Z to get more accurate Z position
G29; EZABL mesh generation
G4 S10; wait for heaters to recover
M117 Purge extruder
G92 E0 ; reset extruder
G1 Z1.0 F3000 ; move z up little
G1 X0.1 Y20 Z0.3 F5000.0 ; move to start-line position
G1 X0.1 Y200.0 Z0.3 F1500.0 E15 ; draw 1st line
G1 X3.5 Y200.0 Z0.3 F5000.0 ; move to side a little
G1 X3.5 Y20 Z0.3 F1500.0 E30 ; draw 2nd line
G92 E0 ; reset extruder
G1 Z1.0 F3000 ; move z up little
M117 EzABL Start
; process Process2
**; layer 1, Z = 0.120**`

`G1 X260.543 Y187.280 F1320
G92 E0.0000
G1 E-6.0000 F3000
**; layer 2, Z = 0.240**
M106 S255
; feature inner perimeter
; tool H0.120 W0.480
G1 Z1.120 F1002`

`G1 X13.434 Y195.684 F2640
G92 E0.0000
G1 E-6.0000 F3000
**; layer 3, Z = 0.360**
; feature inner perimeter
G1 Z1.240 F1002`

and so on...... I cannot seem to figure out how to solve this.

OllisGit, 2021-01-29:
Hi @spankurmonkey,
 
well...there is a missing hint in the description. For paring-performance the "comment layer indicator" must start at the beginning of a line.
Instead of this:
```
G1 X13.434 Y195.684 F2640 G92 E0.0000 G1 E-6.0000 F3000 ; layer 3, Z = 0.360
```
use this:
```
; layer 3, Z = 0.360
G1 X13.434 Y195.684 F2640 G92 E0.0000 G1 E-6.0000 F3000 
```

OllisGit, 2021-01-30:
Hi @spankurmonkey ,

new release 1.25.0 is out and I added a comment.
Please test and give feedback.

Thx, in advance
Olli

spankurmonkey, 2021-01-30:
fixed it.

closed 2021-01-30 by spankurmonkey