OllisGit/OctoPrint-DisplayLayerProgress #198: Duplicate M117 lines With Arc Welder enabled

ssendev, 2021-01-28:
I tried to hack around it by using a negative lookahead regexp but wasn't able to solve it `count	[;BEFORE_LAYER_CHANGE(?!\r\nM117 INDICATOR-Layer.*)]`

OllisGit, 2021-01-29:
Hi @ssendev,

sorry, but I have no idea what you are talking about. Please explain a little bit deeper the issue. In most cases a screenshot is very helpfull ;-)

ssendev, 2021-01-30:
Im using PrusaSlicer so the uploaded G-code contains lines like
```
;BEFORE_LAYER_CHANGE
```
DisplayLayerProgress detects that a file was uploaded and inserts `M117` commands so that the file afterwards contains this
```
;BEFORE_LAYER_CHANGE
M117 INDICATOR-Layer1
```
Then Arc Welder mutates the file which causes DisplayLayerProgress to operate on that file again leaving it like this
```
;BEFORE_LAYER_CHANGE
M117 INDICATOR-Layer1
M117 INDICATOR-Layer1
```

That regexp was a try to ignore `BEFORE_LAYER_CHANGE` lines if they are followed by a `M117` line instead of inserting a second `M117` line. But i'm usually not using python so maybe my regexp is broken or maybe multiline mode is disabled or maybe something else.

The possible solutions i see are either - like i tried - to detect existing `M117` lines and not emitting them again or inserting a `;DisplayLayerProgress` line at the beginning and abort if it's detected

OllisGit, 2021-01-30:
Thx for the detailed explanation ;-)

My solution now looks like this: before adding the M117 INDICATOR the plugin reads the first lines (max 500) and if there is already an M117 INDICATOR, then the processing is stopped.

Hope that helps also for ArcWelder.

OllisGit, 2021-01-30:
Hi @ssendev,

new release 1.25.0 is out.
Please test and give feedback.

Thx, in advance
Olli

ssendev, 2021-01-31:
1.25.0 seems to be broken. After updating it it's not listed as installed. The logs contain:

```
2021-01-31 00:57:01,358 - octoprint.plugin.core - ERROR - Error loading plugin DisplayLayerProgress
Traceback (most recent call last):
  File "/home/ai/OctoPrint/venv/lib/python3.6/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/ai/OctoPrint/venv/lib/python3.6/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/ai/OctoPrint/venv/lib/python3.6/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/ai/OctoPrint/venv/lib/python3.6/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 684, in _load
  File "<frozen importlib._bootstrap>", line 665, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 678, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/ai/OctoPrint/venv/lib/python3.6/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 27, in <module>
    from octoprint_DisplayLayerProgress.CachedSettings import CachedSettings
ModuleNotFoundError: No module named 'octoprint_DisplayLayerProgress.CachedSettings'
```

I tried Installing it again and while it says successfully installed it's still not showing up. Completely restarting didn't help either.

ssendev, 2021-01-31:
1.25.3 works and the doubled lines are gone

closed 2021-01-31 by ssendev

OllisGit, 2021-01-31:
GREAT!!!!