OllisGit/OctoPrint-DisplayLayerProgress #199: Total Height is just reporting the maximum Z of build space

achampion, 2021-01-29:
Currently `total_height` is just the height of my build space.
This does not show the maximum z height with extrusion or use the MAXZ in the Cura gcode.

OllisGit, 2021-01-30:
Duplicated, see #187

OllisGit, 2021-01-30:
Please compare your expected height with the model height from the gcode tab and attach an screenshot to this issue.

I can read the maxz from the gcode-file, but in the past I received a lot of issues that the plugin-height doesn't match with the gcode viewer height.

achampion, 2021-01-30:
I don't know that is a duplicate, as it looks like an off by 1 error vs. the full build height. When I'm back I will post the screenshot.

OllisGit, 2021-02-10:
Hi @achampion,

please look into ```.metadata.json``` file, because DLP use this value [analysis][dimensions][height].

The file is located in the OP upload-folder. This file lists all gcodes with some statistics, e.g.
```
   "example-modifiedByMe.gcode": {
...
        "analysis": {
            "estimatedPrintTime": 4171.766526074045,
            "printingArea": {
                "maxZ": 7.5,
                "maxX": 50.862,
                "maxY": 56.516,
                "minX": -143.0,
                "minY": -74.0,
                "minZ": 0.1
            },
            "dimensions": {
                "width": 193.862,
                "depth": 130.516,
                "height": 7.4
            },
```

achampion, 2021-02-10:
Here's the analysis of a simple nose clip:

```
    "Parameterized_Nose_Clip_Remix_v0_-_Default_Slit.gcode": {
        ...
        "analysis": {
            "analysisPending": false,
            "estimatedPrintTime": 2332.0489274758106,
            "printingArea": {
                "maxX": 204.8209991455078,
                "maxY": 201.02999877929688,
                "maxZ": 330.0,
                "minX": 115.1719970703125,
                "minY": 152.7449951171875,
                "minZ": 0.30000001192092896
            },
            "dimensions": {
                "width": 89.64900207519531,
                "depth": 48.285003662109375,
                "height": 329.69999998807907
            },
```

However the gcode tab in octoprint shows:

```
Model info
Model size: 89.65mm × 48.28mm × 12.60mm
Estimated layer height: 0.15mm
Estimated total print time: 00:21:12
Layers with extrusion: 86
```

Just `12.6mm` in height.

OllisGit, 2021-02-12:
Strange....my assumption was that the model-information of the gcode-tab is the same as the metadata-file.
I need to ask the plugin-dev-community.

closed 2021-03-25 by stale[bot]

OllisGit, 2021-04-04:
Hi @achampion
looks like it is related to this issue. 
https://github.com/OctoPrint/OctoPrint/issues/4036

Fixed in OP1.6.0.