OllisGit/OctoPrint-DisplayLayerProgress #200: v1.25.0 won't install

M1dn1ghtN1nj4, 2021-01-31:
Seems there is something wrong with the v1.25.0 update.  Attempting to update or install it results in "Install: Unknown".  From here, doing a reboot results in an install that is not recognized by OctoPi, not in the installed plugin list, and not accessible through any menu or other plugin (like OctoDash).

However, when attempting to install v1.24.0, it works just fine, with result "Install DisplayLayerProgress Plugin:"

bwinterton, 2021-01-31:
I am also having the same issue on my install. I have also tried to re-install the plugin but get the exact same result every time.

M1dn1ghtN1nj4, 2021-01-31:
The only way (for me) to get out of it, is to flash OctoPi from scratch, and start over.  I'm testing this on a project machine that isn't a production unit, so I don't mind the extra reflash time, but still.

wookie73, 2021-01-31:
Same for me too

Timbo200, 2021-01-31:
> The only way (for me) to get out of it, is to flash OctoPi from scratch, and start over. I'm testing this on a project machine that isn't a production unit, so I don't mind the extra reflash time, but still.
> 
> It might be something that was simply overlooked.

I'm a bit confused. Were you able to install 1.25.0 on a fresh image as new install?

As an upgrade I can't get 1.25.0 to install the same as others here noted.  However, I was able to (re)install 1.24.0 over the failed 1.25 .0 installation.  No need to nuke the existing image.

M1dn1ghtN1nj4, 2021-01-31:
No, I tried to install 25 on a clean install and got the same "unknown" result.  And at least for me, I didn't have any luck installing 24 over the failed 25.  It said it was installed, but it was never available for me.  Might just be me, I have no idea.

PGlyre, 2021-01-31:
Plugin Manager says:
[image]

OllisGit, 2021-01-31:
Hi,

new bugfix version [1.25.1](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.25.1) is out.
Sorry, forgot to add new file to git repository.

BR
Olli

Duplicated, tracked here: #203

closed 2021-01-31 by OllisGit