OllisGit/OctoPrint-DisplayLayerProgress #201: New version  breaks dashboard

KenLucke, 2021-01-31:
After installing the latest version, Dashboard no longer recognizes the Display Layer Progress information
 
[image]

Extra-Fox, 2021-01-31:
Same problem here. It showed up in the plugin managers "clean up" tab. I've tried re-installing it twice now and nothing happens.

Stephan296, 2021-01-31:
You can go back to version 1.24 to use: ```https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.24.0/master.zip``` in the plugin manager

OllisGit, 2021-01-31:
Hi,

new bugfix version [1.25.1](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.25.1) is out.
Sorry, forgot to add new file to git repository.

BR
Olli

Duplicated, tracked here: #203

closed 2021-01-31 by OllisGit

KenLucke, 2021-01-31:
> new bugfix version [1.25.1](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.25.1) is out.

Will try it when prints are done (soonest one is 84 more hours)