OllisGit/OctoPrint-DisplayLayerProgress #202: New version (250) doesn't run

darth-wader, 2021-01-31:
The plug-in does not appear as an installed plugin, even after it's installed.
From the octoprint log:
```
2021-01-30 18:28:18,670 - octoprint.plugin.core - ERROR - Error loading plugin DisplayLayerProgress
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 27, in <module>
    from octoprint_DisplayLayerProgress.CachedSettings import CachedSettings
ModuleNotFoundError: No module named 'octoprint_DisplayLayerProgress.CachedSettings'
```

coehlers, 2021-01-31:
Having same issue

lukemaurer, 2021-01-31:
Same here, only I seem to be getting both that backtrace and another one in my `octoprint.log`:

```
2021-01-31 02:52:00,312 - octoprint.plugin.core - ERROR - Error loading plugin DisplayLayerProgress
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 26, in <module>
    from octoprint_DisplayLayerProgress import stringUtils
ImportError: cannot import name stringUtils
```

BiOs2k, 2021-01-31:
Same Here, I had to revert to version 1.24
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.24.0/master.zip
this is the fix to bring back working

Cha-Doo, 2021-01-31:
You can downgrade to previous version with this link
https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.24.0/master.zip

Steve021171, 2021-01-31:
same problem. Octoprint says a unknown plugin was successfully installed, but the plugin never appears in the plugin manager, also Dashboard doesn't work anymore.
Went back to 1.24 and it works now.

I do not recommend to upgrade to 1.25.

tigercjn, 2021-01-31:
Same Problem. I will have to revert back to 1.44

mpnortn, 2021-01-31:
Same. Steve021171 described what experience as well.

OllisGit, 2021-01-31:
Hi,

new bugfix version [1.25.1](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.25.1) is out.
Sorry, forgot to add new file to git repository.

BR
Olli

closed 2021-01-31 by OllisGit