OllisGit/OctoPrint-DisplayLayerProgress #203: New version won't install

mezdezin, 2021-01-31:
Can't install, tried to install as "unnamed" and isn't actually available upon reboot

yohannaftali, 2021-01-31:
same, after upgrading cannot start

2021-01-31 08:45:14,514 - octoprint.plugin.core - WARNING - Plugin LayerDisplay (0.4.3) is not compatible to Python 3.7.3 (compatibility string: >=2.7,<3).
2021-01-31 08:45:14,576 - octoprint.plugin.core - ERROR - Error loading plugin DisplayLayerProgress
Traceback (most recent call last):
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 1298, in _import_plugin
    module = _load_module(module_name, spec)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/plugin/core.py", line 69, in _load_module
    return imp.load_module(name, f, filename, details)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 238, in load_module
    return load_package(name, filename)
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint/vendor/imp.py", line 212, in load_package
    return _load(spec)
  File "<frozen importlib._bootstrap>", line 696, in _load
  File "<frozen importlib._bootstrap>", line 677, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 728, in exec_module
  File "<frozen importlib._bootstrap>", line 219, in _call_with_frames_removed
  File "/home/pi/oprint/lib/python3.7/site-packages/octoprint_DisplayLayerProgress/__init__.py", line 27, in <module>
    from octoprint_DisplayLayerProgress.CachedSettings import CachedSettings
ModuleNotFoundError: No module named 'octoprint_DisplayLayerProgress.CachedSettings'
2021-01-31 08:45:14,766 - octoprint.plugins.pluginmanager - WARNING - The plugin was installed successfully, but couldn't be found afterwards to initialize properly during runtime. Please restart OctoPrint.
2021-01-31 08:45:16,333 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 4.78ms
2021-01-31 08:45:18,336 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 5.87ms
2021-01-31 08:45:20,338 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 5.89ms
2021-01-31 08:45:22,334 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 4.03ms
2021-01-31 08:45:24,334 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 4.14ms
2021-01-31 08:45:26,339 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 7.78ms
2021-01-31 08:45:28,334 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 5.15ms
2021-01-31 08:45:30,334 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 4.51ms
2021-01-31 08:45:32,337 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 5.91ms
2021-01-31 08:45:34,333 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 3.99ms
2021-01-31 08:45:36,336 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 5.41ms
2021-01-31 08:45:38,342 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 9.99ms
2021-01-31 08:45:40,333 - tornado.access - WARNING - 404 GET /plugin/DisplayLayerProgress/values (::ffff:192.168.1.30) 4.18ms

OllisGit, 2021-01-31:
Hi,

new bugfix version [1.25.1](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.25.1) is out.
Sorry, forgot to add new file to git repository.

BR
Olli

Beo666, 2021-02-02:
Works greate, thank you.

mezdezin, 2021-02-02:
Yup fixed thank you

closed 2021-02-02 by OllisGit