OllisGit/OctoPrint-DisplayLayerProgress #204: Can't retrieve layer progress! After after update to 1.25.1

Boxter83, 2021-01-31:
[image]

sebbl2, 2021-01-31:
Same here (Prusa MK3s)

OllisGit, 2021-01-31:
Hi @sebbl2 ,

thanks for reporting. It is now fixed in Version 1.25.2

BR
Olli

adrensnyder, 2021-01-31:
I've updated to 1.25.2 but same error

sebbl2, 2021-01-31:
For me it worked. Although i deleted the previous version Incl.data

adrensnyder, 2021-01-31:
> For me it worked. Although i deleted the previous version Incl.data

Where is this file? Thanks

sebbl2, 2021-01-31:
What do you mean. Uninstall it in Octoprint. Afterward reinstall it. It might be shown in the plugin manager as 1.25.0/1 but after its de-zipped its 1.25.2

sebbl2, 2021-01-31:
it will download the latest master, wich is 1.25.2

adrensnyder, 2021-01-31:
> What do you mean. Uninstall it in Octoprint. Afterward reinstall it. It might be shown in the plugin manager as 1.25.0/1 but after its de-zipped its 1.25.2

I've already updated to 1.25.2 but same error. Now i try to remove completely and reinstall it

adrensnyder, 2021-01-31:
Nope. I've removed the plugin and clean data. Reinstalled and reuploaded the gcode and no Layers found:
[image]
[image]
[image]

What i'm doing wrong?

sebbl2, 2021-01-31:
I did not change any settings either. Did you transfer the file to OP with the the buggy plugin installed ? Maybe just delete the g-code-file and transfer it again.

adrensnyder, 2021-01-31:
> I did not change any settings either. Did you transfer the file to OP with the the buggy plugin installed ? Maybe just delete the g-code-file and transfer it again.

I've created another gcode and i've uploaded it after remove and reinstall

sebbl2, 2021-01-31:
Ok. My fault.....isnt working here neither. The pre-update g-codes still show layer progress. Since update same error as @adrensnyder. Sorry just recognized yet.

sebbl2, 2021-01-31:
@adrensnyder 
meanwhile you can go back to 1.24 
`pip install "https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.24.0/master.zip"`
or
`~/oprint/bin/pip install "https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.24.0/master.zip"`
or

adrensnyder, 2021-01-31:
> @adrensnyder
> meanwhile you can go back to 1.24
> `pip install "https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.24.0/master.zip"`
> or
> `~/oprint/bin/pip install "https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.24.0/master.zip"`
> or

Thanks for this. I was searching for it 😃

neilbunn, 2021-02-02:
I'm still getting the same error as above, using Prusa Slicer 2.3, latest Octoprint, and the 1.25.3 version, also using a Prusa MK3S. Did something change in how Prusa comments the code to trigger this? Unfortunately I updated the slicer, and Octoprint all on the same day! <laugh>

OllisGit, 2021-02-02:
Hi @neilbunn,

please provide the octoprint.log and also the file directly generate by your slicer, so that I am able to reproduce the issue with the latest version of DLP.

neilbunn, 2021-02-02:
> Hi @neilbunn,
> 
> please provide the octoprint.log and also the file directly generate by your slicer, so that I am able to reproduce the issue with the latest version of DLP.

Attached. Thanks for all you do an a great plugin, I've loved it so I hope this helps with this error. May entirely be on the user side though, but I haven't done anything other than the Prusa Slicer upgrade and the Octoprint/DLP upgrade in the last week. Was working fine before that.

[octoprint.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5914856/octoprint.log)
[Arc-Reactor-Ring1-Ring2-Top-Battery_0.15mm_PLA_MK3S_8h50m (1).gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5914858/Arc-Reactor-Ring1-Ring2-Top-Battery_0.15mm_PLA_MK3S_8h50m.1.gcode.zip)

phealy, 2021-02-04:
I suspect this might be related - when I upload a file directly into a folder (using foldername/filename.gcode in PrusaSlicer), the actual gcode goes into the correct place, but an additional file with the comments from this plugin ends up in the root of the uploads folder. This is on OctoPrint 1.5.3 running with Python 3.9.1 and plugin version 1.25.3.

[prusaslicer_export.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5925709/prusaslicer_export.txt)

[file_in_upload_root.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5925707/file_in_upload_root.txt)

[processed_file_in_folder.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5925708/processed_file_in_folder.txt)

OllisGit, 2021-02-07:
Hi @phealy, this is related to issue #208, #207

Daskew78, 2021-02-22:
Having the same issue here also. Had to revert back.

OllisGit, 2021-02-28:
Hi @Daskew78,

which version of DLP do you use? 

Hi @phealy,
is it fixed for you?

achampion, 2021-03-03:
Continues in 1.25.4.  
OctoPrint 1.5.3. 
Cura slicer 4.8. 

Looking at the gcode created by Cura shows:
```
...
;LAYER_COUNT:348
;LAYER:0
M107
...
```

phealy, 2021-03-04:
> Hi @phealy,
> is it fixed for you?

It's resolved, thanks!

Daskew78, 2021-03-04:
> Hi @Daskew78,
> 
> which version of DLP do you use?
> 
> Hi @phealy,
> is it fixed for you?

Hi, I was using 1.25.1 but had to revert back for now.

LightningShark25, 2021-03-24:
New user to DLP.  Experiencing the same issue.  
DLP  1.25.4

Here is a snippet from my GCODE
G1 X186.267 Y36.263 E0.00461 ; infill
;LAYER_CHANGE
;Z:0.8
;HEIGHT:0.2
;BEFORE_LAYER_CHANGE
G92 E0
;0.8

Here is my DLP layer change patterns...
[image]

OllisGit, 2021-04-04:
Hi @LightningShark25,

please provide the full gcode and also the expression textarea-content instead of an image. With the image I could not reproduce the issue, btw. did you modify the default expression?

Hi @Daskew78,
which version of DLP is working for you?

LightningShark25, 2021-04-07:
Hi sorry for the delay.

I had not modified the settings no. They were as standard
HOWEVER it is now working.

As far as I can tell I didn't change anything so not sure what happened

I had restarted the octoprint a couple of times when it wasn't.

Thanks for your time

closed 2021-05-18 by stale[bot]