OllisGit/OctoPrint-DisplayLayerProgress #205: Display layers problem after update

bgiovanny, 2021-01-31:
Hello, after last update I got this message:

Attention
Layer indicator not found in file: 'Mount_).gcode'
Check layer pattern in DisplayLayerProgress-Settings and reupload the file!

I use IdeaMaker how to fix it?

tks

Stoffl1981, 2021-01-31:
Me also, but the file has the ";Layer" indicator in it

JasonRS1969, 2021-01-31:
Experiencing the same issue here.

Octoprint 1.5.3
DisplayLayerProgress plugin version 1.25.2

Gcode generated by cura (4.8) and uploaded through Octoprint interface.

GCode files from Cura contain ;Layer indicator but do not have the M117 INDICATOR-Layer added on upload. (GCODE in txt below)

[Zit Test Print - Draft.gcode.txt](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/5899725/Zit.Test.Print.-.Draft.gcode.txt)

Was working prior to upgrade to 1.25.2 (skipped 1.25.0, so issue may be present there)

CaptainSweatpants, 2021-01-31:
Came here to report the same error. 
All newly uploaded files report this error; even if uploading the exact same file before the update..
Gcode files uploaded ad stored before the update continue to work fine.

domhardt, 2021-01-31:
I have the same problem.

Whitehawk2000, 2021-01-31:
Same problem here after the update.

Octoprint 1.5.3
DisplayLayerProgress V1.25.2
PrusaSlicer-2.3.0+

ddpt83, 2021-01-31:
Same problem after the update.
Octoprint 1.5.3
DisplayLayerProgress V1.25.2
PrusaSlicer-2.3.0+win64

Spacekoaster, 2021-01-31:
Hello, same problem here after the latest update:
Octoprint 1.5.3
DisplayLayerProgress V1.25.1
PrusaSlicer V2.3.0 with the ;LAYER:[layer_num] in the before layer change gcode

rmknightz, 2021-01-31:
Hi,

Same issues.

Octoprint 1.5.3
DisplayLayerProgress V1.25.2
Cura V4.8 - Mac

KRUP85, 2021-01-31:
Hi,

same here 
Octoprint 1.5.3
DisplayLayerProgress V1.25.1
PrusaSlicer V2.3.0 with the ;LAYER:[layer_num] in the before layer change gcode

Pampinho1, 2021-01-31:
Ditto... New version doesn't seem to be matching right with the Regex configured to identify the ";LAYER:XXX" in the uploaded GCODE. The Slicers ARE including the layer indicator strings in them, but the plugin doesn't seem to find them. For the time being I will remove it as I don't know enough of this code play with it.

adrensnyder, 2021-01-31:
Same as https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/204

OllisGit, 2021-01-31:
Sorry about the short release interval and the open issues!

Latest Version 1.25.3 fix the issue.

Btw. some comments were not very helpful, next time please always provide:
- octoprint.log
- used gcode
- screenshot

Thx, in advance
Olli

djlightflighter, 2021-01-31:
> Sorry about the short release interval and the open issues!
> 
> Latest Version 1.25.3 fix the issue.
> 
> Btw. some comments were not very helpful, next time please always provide:
> 
> * octoprint.log
> * used gcode
> * screenshot
> 
> Thx, in advance
> Olli

Sorry bro thank for fixing it  and where is the coffee contribution link ?  i got you covered

OllisGit, 2021-01-31:
Hi @djlightflighter,
 you ask for, here it is:

https://www.paypal.com/donate/?cmd=_s-xclick&hosted_button_id=6SW5R6ZUKLB5E&source=url

djlightflighter, 2021-01-31:
Yep I just sent it to you !!!!!

jk257, 2021-02-01:
Having the same issue after upgrading to python 3 today. Any new gcode file I upload to octoprint is not getting the M117 lines added by DLP like it did for all other files I've added before today. I just uploaded a file sliced with the same version of Cura, no other changes than upgrading to P3 and installing a few other plugins.

OllisGit, 2021-02-01:
Hi @jk257,

which version of OctoPrint and the plugin do you use?

Please provide the octoprint.log and also the uploaded gcode file. Otherwise I can't analyse.

JasonRS1969, 2021-02-01:
Confirming that  1.25.3 has resolved the issue for me. 

Have tested several Gcode files and all seems well so far.

KenLucke, 2021-02-01:
1.25.3 seems to have solved the problem for me.

ddpt83, 2021-02-01:
Version 1.25.3 has resolved the problem for me.

gstevenson, 2021-02-01:
How do we force an upgrade? I'm on 1.25.2 and suffering the problem but OctoPrint doesn't think I've got any updates to install

gstevenson, 2021-02-01:
1.25.3 also confirmed fixed for me 👍

djlightflighter, 2021-02-02:
Copy the url into the get more plug-ins space

closed 2021-02-02 by OllisGit