OllisGit/OctoPrint-DisplayLayerProgress #207: File generate during upload

Jeggo65, 2021-02-02:
Hello,

first of all the plugin is working :-)
When I upload a file to OctoPrint via Pursa Slicer 2.3 (e.g. local OctoPrint drive: Test/Test_Cube.gcode), a file in the root directory with the same name is created and contains this:

; BEGIN DISPLAYLAYERPROGRESS SETTINGS
; DisplayLayerProgress_layerIndicatorProcessed = true
; END   DISPLAYLAYERPROGRESS SETTINGS

In my setup (latest OctoPi and OctoPrint versions), the file remains in the root directory after the print is finished.
Could this file (automatically) be deleted after the print or even before if the gcode is scanned and processed?
Or if this is not possible could the file be generated in an extra directory (e.g. DLP)?

CU

Jens

OllisGit, 2021-02-02:
Thx, for reporting!
I never tested the function to upload an gcode to an sub-folder.
I think I only used the filename instead of the full-path.

I will take a look on this.

Jeggo65, 2021-02-02:
Maybe it is just a good idea to hide these files or put them in a folder, that is not shown in OctoPrint. When you select this file, you will get an error, that no layer information in this file. As far as I understand it, these files do not contain any data, that has to be edited by the user.

HedgesHandmade, 2021-02-06:
I'm experiencing the same issue as it relates to the DeleteAfterPrinting plug-in. I have it set to move the file to the _archive folder after successfully printing, which it does successfully, but it also leaves the same 130 byte file with the same filename in the main file list. The text inside the file reads:

```
; BEGIN DISPLAYLAYERPROGRESS SETTINGS
; DisplayLayerProgress_layerIndicatorProcessed = true
; END   DISPLAYLAYERPROGRESS SETTINGS
```

OllisGit, 2021-02-10:
Hi! 
Fixed in lates release V1.125.4

Please test and give feedback.

Thx, in advance!
Olli

Jeggo65, 2021-02-10:
Thanx Olli,

it is working. Text lines are added to the original file.

CU

Jens

closed 2021-02-10 by Jeggo65