OllisGit/OctoPrint-DisplayLayerProgress #208: Cut and Paste Leaves Almost-Empty File in Original Location

cr45hmurphy, 2021-02-06:
Salandora/OctoPrint-FileManager#37

HedgesHandmade, 2021-02-06:
I'm experiencing the same issue as it relates to the DeleteAfterPrinting plug-in.  I have it set to move the file to the _archive folder after successfully printing, which it does successfully, but it also leaves the same 130 byte file with the same filename in the main file list.  The text inside the file reads:

```
; BEGIN DISPLAYLAYERPROGRESS SETTINGS
; DisplayLayerProgress_layerIndicatorProcessed = true
; END   DISPLAYLAYERPROGRESS SETTINGS

```

OllisGit, 2021-02-07:
similar to #207

OllisGit, 2021-02-10:
Hi @cr45hmurphy! 
Fixed in lates release V1.25.4

Please test and give feedback.

Thx, in advance!
Olli

cr45hmurphy, 2021-02-11:
> 
> 
> Hi @cr45hmurphy, @HedgesHandmade !
> Fixed in lates release V1.25.4
> 
> Please test and give feedback.
> 
> Thx, in advance!
> Olli

Updating now. Will provide feedback shortly.

cr45hmurphy, 2021-02-11:
It appears to be fixed!

closed 2021-02-11 by cr45hmurphy

HedgesHandmade, 2021-02-11:
> Hi @cr45hmurphy, @HedgesHandmade !
> Fixed in lates release V1.25.4
> 
> Please test and give feedback.
> 
> Thx, in advance!
> Olli

Working perfectly now!