OllisGit/OctoPrint-DisplayLayerProgress #209: Printer Display, conditional if this, then that

lightmaster, 2021-02-11:
This possibly already exists, but I can't seem to figure out how to do it if you can. I'd like to have conditional statements in the textbox used to generate the `M117` codes for the printer display. Something like `if there is an upcoming filament change, then display time left until filament change; else display estimated print completion time`. I see the variables in red that would be needed to accomplish this, but I can't figure out how to do the conditional statement, if that's event possible right now.

OllisGit, 2021-02-12:
Hi @lightmaster,
this is currently not possible. you set the variable in the output-pattern --> the value is shown.

I like the idea, but the first question that comes to my mind is: how should be the userinterface to handle all combinations...maybe on the left side list all events that could appear and on the right side you allign such event with a text base condition.....

....do you have a suggestion about that?

lightmaster, 2021-02-13:
I was thinking something much similar, all text based within a larger text box. Something like:
```
{if [changefilamenttime_left] != ""} Filament change in [changefilamenttime_left] {else} Done in [printtime_left] {endif}
```

In this example, I'm not sure what `[changefilamenttime_left]`'s null value would actually be, but it would be adjusted such that it's only true if there is a filament change remaining.