OllisGit/OctoPrint-DisplayLayerProgress #210: Add Bed and Extruder temperature placeholders for messages

sbussinger, 2021-02-25:
It would be really nice if there were bed and extruder temperature keywords in the list of placeholders used in messages. I'd like to have the temperatures in the NavigationBar and would prefer not to install another plugin to do it when I'm using DisplayLayerProgress to modify the NavigationBar already.

OllisGit, 2021-04-04:
Hi @sbussinger,

there is no server-side event for the temp-change, so I need to implement a timer for reading the current temp.

How often do you need a temp update in the UI e.g. is every 10 seconds enough?

sbussinger, 2021-04-04:
Every 10 seconds would be fine. A configuration setting would be nice I suppose, but not important for me. If I want real-time readings I can go to the temperature tab. But if adding timer support to the plugin is a big deal, I'd say skip this request. I thought perhaps it'd be easy to add along side existing stuff, if it's not simple then this isn't worth your time.

Thanks!

OllisGit, 2022-01-26:
Hi @sbussinger,

in the latest version 1.28.0 you can output the temperatures where ever you like ;-)

Please test and give feedback.

Thx,
Olli

sbussinger, 2022-01-27:
That's awesome, thank you very much! This is great!

Feedback: 
1) Perhaps limit the temperature displays to 1 decimal point accuracy instead of 2 to match the numbers shown on the Temperature tab. Also the 0.01° resolution causes the number to fluctuate more than it needs to.
2) I have enabled the option to append the actual bed temperature to the browser tab title and it's blinking on and off erratically (I don't recall this behavior before).

Again, thank you. It's much appreciated!

OllisGit, 2022-03-23:
Hi @sbussinger,
the first point will be in the next release.

The second point I am not able to reproduce.
- What browser do you use?
- What DLP-Settings do you use? Is the blinking invervall like the "Toggle Printer Display Message" time?

sbussinger, 2022-03-23:
On the second point, I normally use Edge, but it does the same thing on Firefox. I'll attach a short screen video showing the effect. It's not a regular pattern of blinking as far as I can tell, it's more random.

https://user-images.githubusercontent.com/706838/159787125-63b79083-4a39-466c-89de-d2074d265342.mp4

I've had the Output Interval set to 0. I tried setting it to 5 seconds and it had no obvious effect, so I've set it back to 0. I don't have an LCD on the printer so I don't have the printer display stuff enabled.

This is not a big deal and not worth spending a lot of time on.

OllisGit, 2022-07-11:
Hi @sbussinger,
thanks for the video was very helpful.

My assumption is that reading all temperatures from the printer that sometimes the bed-temperature is not always in the "response". So, I now store the last bed-temperature and if the current temperature could not read, I display the last readed temperature instead.

This is implemented and now available in version 1.29.0rc1. 

If you are using OP 1.8.x you can switch between the release-channels via "Software Update" menu.
Or you can reinstall the version via plugin manager: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.29.0rc1/master.zip

Please test and give feedback.

Thx, in advance
Olli