OllisGit/OctoPrint-DisplayLayerProgress #214: [FR] Add new placeholder for the file name

Majestry, 2021-03-12:
It will be a nice feature to add new placholder, which will be replaced with the file name when sending to the printer.
For example, something like this
_[current_print_filename]_
Then the output the printer can be like  progress is 
_[current_print_filename] progress is [progress]% L=[current_layer]/[total_layers]_

OllisGit, 2021-04-04:
Hi @Majestry,
fyi: will be in the next release. 

BUT if you use this parameter in "Print Display", the M117 message could be really long. Be aware of this:
[image]

Majestry, 2021-04-04:
Cool! Thank you a lot! 
Yes, i know about the long texts. Hoperfully, my printer just shows 64 symbols and simply cut other text. (Ender 3 v2 with Jyers firmware)

OllisGit, 2021-04-05:
Hi @Majestry 

new release 1.26.0 is out, please test and give feedback.

Thx, in advance
Olli

Majestry, 2021-04-08:
Yes, it works.
Checked with latin file names
Printer display message in settings: 
_[current_print_filename] Layer [current_layer]/[total_layers]. Heigth [current_height]/[total_height]mm._
Printer display output: 
[image]

closed 2021-04-12 by Majestry