OllisGit/OctoPrint-DisplayLayerProgress #215: Add support for @pause

sbussinger, 2021-03-16:
You're currently detecting M600 commands to recognize an upcoming filament change. Any chance of adding detection of `@PAUSE` commands to recognize an upcoming print pause? OctoPrint documentation for `@PAUSE`: 

https://docs.octoprint.org/en/master/features/atcommands.html?highlight=pause

OllisGit, 2021-04-04:
hmm...just for my understanding: you want to add ```@pause``` in your gcode instead of ```M600```!?!

So, the implementation should look for ```M600``` or```@pause```. In both cases the plugin should act the same, right?

sbussinger, 2021-04-04:
Yes, handling it exactly like M600 is fine. As a workaround I've been putting both an M600 and @pause into the gcode. That's worked because my printer happens to ignore M600 commands, but my future printers are likely to support M600. Even better would be changing the message to "next pause" instead of "next filament change", but honestly it really isn't necessary.

Thanks!

OllisGit, 2021-04-05:
Hi @sbussinger,

I looked into the handling of @-commands and it looks like that it is currently "incompatible" with my implementation or in other words: it took a lot of time to implement this feature.

So, because you currently don't have a "real demand" for that feature, please come back later (create new issue) if your new printer has some trouble handling the M600 command.

Sorry,
Olli

closed 2021-04-05 by OllisGit

sbussinger, 2021-04-05:
Got it - that makes sense. My workaround is good for now. Thanks for taking a look at it!