OllisGit/OctoPrint-DisplayLayerProgress #216: DisplayLayerProgress_feedrateChanged event spam when connected to MQTT

carnifex82, 2021-03-16:
I have the official MQTT Plugin running to report octopi status to my raspberry openhab installation via mqtt. After I installed the DisplayLayerProgress Plugin, I noticed that the Updates on the raspberry started lagging a bit behind, especially after a layer change. 

At first I thought that the octopi (on it's own raspberry) can't handle the plugin, but when I connected to the mqtt broker and subscribed to the octoPi topic, I realized that this plugin is creating a lot of new events in MQTT. In general, this is very great, because there is a lot of more information in those events (for example: heightChanged) than octopi offers otherwise over mqtt. 

However, there is a lot of spam of DisplayLayerProgress_feedrateChanged events, that's taking a tax on the openhab server in processing the event channel. 

I found the option to set an update interval and set this to 30s. However, this is only for the DisplayUpate, it has no affect on the mqtt messages. 

I'm not even sure if you generate those mqtt messages or if they are just generated from octopi and the official mqtt plugin, based on the events of this plugin.

However, it would be great if it is possible to keep the mqtt events, but choose which ones to get, so that you can unsubscribe from the feedrateChanged event for weaker machines.

OllisGit, 2021-04-04:
Hi @carnifex82,

with "official-plugin" you mean this one: https://plugins.octoprint.org/plugins/mqtt/ ?

Currently, each time a parameter is changed, an DLP-Event is fired on the Octoprint-Eventbus. DLP doesn't generate mqtt-messages. 
So, each other plugin can listen on this OP-Eventbus. My assumption is, that your mqtt-plugin re-publish this 1:1 via mqtt.
But you should be able to just subscribe to the event/topic you really need and ignore all others.

Maybe I didn't get your point.

carnifex82, 2021-04-04:
Yes this one. On the subscriber side, I can ignore those messages, yes. But the the mqtt server needs to handle them all.

If I understood you correctly, my theory seems to be somewhat right. You create events and the plugin publishes them all. You can disable "other events" in the config, but this disables other interesting events as well.

I assumed that the rate limitation should limit the events as well, that's why I created the issue. But I have no idea of the internals, so maybe it's supposed to work like this.

OllisGit, 2021-04-05:
Hi @carnifex82,

I looked into the sourcecode of the plugin and it looks like that there is already a filter-function implemented.
Of course you can disable the "Unclassified events",  but then you didn't receive any DLP-Event.

I think the best approach is to ask for a feature request for the mqtt-plugin. Should be "easy' to implement:
- add a textfield in mqtt plugin-settings where you can add an regular expression for excluding (or including) event-keys:
e.g. exclude feedrate and progress: ```DisplayLayerProgress_feedrateChanged, DisplayLayerProgress_progressChanged```

Please mention the issue here, so that both tickets "alligned"

closed 2021-05-16 by stale[bot]