OllisGit/OctoPrint-DisplayLayerProgress #223: Problem with MAXZ

Neokamikaze, 2021-04-17:
Hello,

I have a problem with MAXZ value. This don't read in interface but the value is in the gcode file. 
**Example:** 
;MINZ:0.2
;MAXX:122.872
;MAXY:120.044
;MAXZ:25.4
;POSTPROCESSED
;Generated with Cura_SteamEngine 4.8.0
M82 ;absolute extrusion mode

Thanks

[image]
[image]

Neokamikaze, 2021-04-18:
Hello, 
Fixed using Print Time Genius Plugin
Ciao

Doormat1, 2021-05-24:
I'm having the same issue. How does print time genius fix this?

OllisGit, 2021-05-25:
@Doormat1  Good question!
@Neokamikaze, please answer!

@Doormat1, please provide a sample gcode where the maxz was not detected. Btw. the total height is provided by OP itself.
When the value is not detected, please take a look into the op-metafile, located in the uploads-dir of op-installation.
The ```.metadata.json``` should include this section:
[image]

Doormat1, 2021-05-25:
None of my gcodes have anything in the analysis part of the json.

Doormat1, 2021-05-25:
I guess Display Layer Progress requires Print time genius to work properly, as Ive run M503 on my printer and now Im getting a star next to g code files and Im getting max z height.

OllisGit, 2021-05-25:
Hi @Doormat1,

PTG is not needed ! 

Did you de-activate ('Never" selected) the analyse feature in OP-Settings? That could be the reason where there is no metadata in the file.

[image]

closed 2021-07-05 by stale[bot]