OllisGit/OctoPrint-DisplayLayerProgress #224: Not possible to use 24h time format?

hapklaar, 2021-04-18:
The plugin only shows am/pm time format for me, I would like to have this in 24h notation.

[image]

In configuration I have this set this which should give me 24h format right?:

[image]

OllisGit, 2021-05-10:
@hapklaar time format option for the current time will be in the next release.

hapklaar, 2021-11-14:
I haven't been able to test as I've since moved to Klipper + Fluidd. As soon as I have the chance I will though and report back if there are issues. Will close for now.

closed 2021-11-14 by hapklaar