OllisGit/OctoPrint-DisplayLayerProgress #226: "Layer Indicator Not Found In File" after slicing through Octoprint Slic3r Plugin

Delphant, 2021-04-21:
Hello,

I'm trying to get the layer indicator to work for the slic3r plugin. after slicing through octopi, I get the message "layer indicator not found in file" I believe that maybe the file is not being seen by the layerprocess plugin after it finishes slicing. I am using a profile made in prusaslicer. I've added the ;LAYER:[layer_num] to see if that caused it, but still nothing.

;BEFORE_LAYER_CHANGE
G92 E0.0
;[layer_z]
;LAYER:[layer_num]

Thanks.

wanderling, 2021-05-10:
I also have been having this same problem forever. Interesting thing is, it complains about no layer information but then seems to operate normally. This seems to point to something in the check code... Attached is a typical file uploaded from Cura over local network to the pi. The file was also auto processed with 'Arc-Welder'.
[CE3_Sudo-bearings.aw.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/6452002/CE3_Sudo-bearings.aw.zip)
[plugin_DisplayLayerProgress_events.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/6452020/plugin_DisplayLayerProgress_events.log)

Delphant, 2021-05-10:
Hello, I believe the plugin-expressions are untouched. I've included a sample gcode sliced within octoprint Slic3r (1.1.7), an image of the expression settings, and my config file. I'm not sure where in the gcode it would say if there is a number in place of the layer, I don't think there is but I'm not sure. Thanks!

[sample gco, expression image, config.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/6453313/sample.gco.expression.image.config.zip)

OllisGit, 2021-05-25:
Hi @Delphant,
your gcode file-extension is not ```gcode``` is that only in the attached zip or did you always don't use gcode. Maybe this is the issue.

OllisGit, 2021-05-25:
Hi @Delphant,

thanks for providing the all the data, really helpfull.

I found the issues:
1.) your gcode doesn't include one of the default-expressions. It contains AFTER_LAYER or layer_z lines.So, you need to enter this line to your expression text-box:
```
count	[;AFTER_LAYER_CHANGE.*]
```
2) you used a downloaded gcode from OP and in this gcode DLP-Plugin added a marker that the file is already processed. And a reprocessing is not done again after upload (complicated to explain). Please look into the file and remove the last three lines:
```
; BEGIN DISPLAYLAYERPROGRESS SETTINGS
; DisplayLayerProgress_layerIndicatorProcessed = true
; END   DISPLAYLAYERPROGRESS SETTINGS
```
After you did the 2 steps, re-upload the file and now the Layer should be detected!

OllisGit, 2021-05-25:
Hi @wanderling,

if you turned off the ArcWelder-Plugin is DLP working fine with the default expressions (do a reset)?

wanderling, 2021-05-29:
> 
> 
> Hi @wanderling,
> 
> if you turned off the ArcWelder-Plugin is DLP working fine with the default expressions (do a reset)?

Here is another file freshly sliced just today... Same result.
[CE3_Bottle_dryer(1).zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/6564043/CE3_Bottle_dryer.1.zip)
[plugin_DisplayLayerProgress_events.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/6564044/plugin_DisplayLayerProgress_events.log)

OllisGit, 2021-06-04:
Hi @wanderling,

I uploaded your gcode and it works fine with the default settings
[image]
 
In the eventlog I can see that arcwelder was still active because the file included this message:
```
'currentFilename': u'CE3_Bottle_dryer(1).aw.gcode',
```

Please try with again with disabled ArcWelder
- disable ArcWelder
- restart the server
- upload your new sliced gcode
- select the code
- the max-layer count should be visible

wanderling, 2021-06-06:
Still persists after disabling arcwelder from settings and restarting.
[plugin_DisplayLayerProgress_events (1).log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/6604649/plugin_DisplayLayerProgress_events.1.log)
[CE3_tmp_1622999419838.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/6604650/CE3_tmp_1622999419838.zip)

wanderling, 2021-07-05:
Looks like something in your plugin sticks around after updates. Completely uninstalling your plugin, restarting octoprint, and reinstalling your plugin from the repository seems to fix the issue. You should add this as a fix in your notes.

Delphant, 2021-07-08:
Hello, sorry for the late response.

I've added the `count	[;AFTER_LAYER_CHANGE.*]` to the profile in the After layer change G-code box, but after manually deleting the lines from step 2, I still get the error when reuploading it. Is there a setting I need to change in the plugin settings that I may be missing? Right now I just have the default settings with 'Add Layer-Indicators during upload' checked and 'Exclude files from layer detection after file selection' unchecked. Thanks!
[xyzCalibration_cube (1).zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/6786970/xyzCalibration_cube.1.zip)

OllisGit, 2021-10-14:
Hi @Delphant,
now your gcode is totally broken ;-)

The DLP-Expression should not be included into your gcode
[image]

The plugin is reading the gcode and detects comment-lines (see https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works).
So in your initial case, it should count theses line ```;AFTER_LAYER_CHANGE``` and because the line is not in the Plugin settings, you need to change the settings to this:
[image]

OllisGit, 2021-10-14:
Hi @wanderling,
I uploaded your model with the default settings (plus layeroffset of one) and the total layer count of 24 was shown:

[image]
 
Please download the gcode file directly from OP. It should include lines like this:
```
M117 INDICATOR-Layer10
```

closed 2021-11-24 by stale[bot]