OllisGit/OctoPrint-DisplayLayerProgress #228: [total_height] not accurate for layer heights other than 0.2 mm [CURA]

Atomique13, 2021-05-02:
Hello

[total_height] is not accurate for layer heights other than 0.2 mm. It shows 6.62 mm instead of 6.72 mm for 0.32 layer height or 7.08 mm instead of 6.96 (I tried on others prints and the same problem). MAXZ in gcode, after DLP post-processed it, shows **;MAXZ:6.72** or **;MAXZ:6.96**, which are good. I added +1 layer offset in Layer settings because CURA adds +1 layer for purge line on the left side, if that helps. I am using CURA 4.9.0, Octoprint 1.6.0 and Creality Ender 3 V2 . Thank you

[image]
GCODE 0.08 Layer Height

[image]
GCODE 0.2 Layer Heigh

[image]
GCODE 0.32 Layer Height

[image]
Octoprint 0.08 Layer Height

[image]
Octoprint 0.2 Layer Height

[image]
Octoprint 0.32 Layer Height

OllisGit, 2021-05-10:
Hi @Atomique13 ,

DLP reads the total height from Octoprints Metafile. Please take a look
``` <op-installation>/uploads/.metadata.json```
Example:
[image]

Maybe there is something wrong in OP.

Atomique13, 2021-05-19:
Hello again,

[image]
.metadata.json 0.08 Layer Height

[image]
.metadata.json 0.2 Layer Height

[image]
.metadata.json 0.32 Layer Height

closed 2021-06-28 by stale[bot]

Sekisback, 2021-09-17:
still open DPL calculate one Layer above MAXZ