OllisGit/OctoPrint-DisplayLayerProgress #230: Fire event when M600 occurs

schmjop, 2021-06-03:
It would be great if an event of DLP was fired when gcode M600 is occurs. This would allow to send a notification or something similar.

OllisGit, 2021-06-19:
Hi @schmjop,
I added M600 event (next reales), so that you can listen on.

What is your plan about receiving this event?

schmjop, 2021-06-21:
Awesome, thank you!

I have Home Assistant connected to OctoPrint using the [OctoPrint-HomeAssistant plugin](https://github.com/cmroche/OctoPrint-HomeAssistant) which allows me to listen to OctoPrint events. Now when a M600 event is detected I am able to send a notification or TTS to a speaker.

OllisGit, 2021-10-14:
Hi @schmjop,

the new release 1.27.0 with this feature is out now.
Please test and give feedback.

Thx, in advance
Olli