OllisGit/OctoPrint-DisplayLayerProgress #231: estimated_end_time showing wrong time

jsdokos, 2021-06-21:
While my print was running I noticed a wrong time being calculated for estimated_end_time.

Current time is reported as: ~2:40
printtime_left is reported as ~1 hour 25 minutes
estimated_end_time is 10:08pm

I am not sure where the extra 6 hours is coming from. The time is reported correctly on the [Dashboard]{https://github.com/j7126/OctoPrint-Dashboard} which uses a lot of the data from your plugin. Let me know if you would like any extra data.

jsdokos, 2021-06-22:
Looks like the difference is from the assigned time zone of the finishing time, and my time zone. BST vs CST

jsdokos, 2021-07-01:
I'm an idiot and didn't see what the Pi was set to, I fixed it and everything works now.

closed 2021-07-01 by jsdokos

magnjo, 2021-08-28:
Sorry, is this "assigned time zone of the finishing time" a setting I can't find?
(Pi time is 08:45 CEST, TabTitle says 84% 08:00 (printing))

jsdokos, 2021-08-28:
I believe I had to change the time zone in the raspi-config

sudo raspi-config
Go to 5 localisation Options > L2 Timezone 
Make sure it's correctly set

I believe that's what I did, but i'm not sure.

magnjo, 2021-08-29:
Thank you, tried that:

pi@octopi:~ $ sudo raspi-config
[sudo] password for pi: 

Current default time zone: 'Europe/Stockholm'
Local time is now:      Sun Aug 29 10:47:56 CEST 2021.
Universal Time is now:  Sun Aug 29 08:47:56 UTC 2021.

Then I started a print. And at

pi@octopi:~ $ date
Sun 29 Aug 11:34:14 CEST 2021

OctoPrint says

Print Time Left: 40 minutes ●

But TabTitle says

52% 11:13 (printing)

So for some reason DisplayLayerProgress thinks current time is ~10:33 which matches neither UTC nor local time... =7 No idea what I could be missing.