OllisGit/OctoPrint-DisplayLayerProgress #242: estimated_end_time not showing 10's of hours

jjaakk, 2021-10-17:
I have DisplayLayerProgress 1.27.1 installed on OctoPrint 1.7.0, OctoPi 0.18.o - just recently updated to use Python 3.

The estimated_end_time shows finish time as 06:49, when it should be showing 16:49. Why is it not showing the 10's of hours ?
The Dashboard Plugin shows ETA with the correct hour and mins, but the DisplayLayerProgress ETA is missing the 10's value.

Hopefully this is just a time setting in DLP plugin.
Jak.
[image]

OllisGit, 2021-10-18:
Hi @jjaakk,
hmmm..did you changed the Timeformat in the Settings?
[image]
Please also activate the normal clock in the NavBar as a reference
[image]

Btw. the Dashboard-Plugin use the javascript browser date function and DLP use the python server time
DashBoard
[image]
DLP-Serverside
[image]

jjaakk, 2021-10-18:
I did change the time format to %I:%M. No difference.
I have activated the normal clock as a reference. No difference.
Thanks for the explanation of the two different time mechanisms.

I think I will just keep an eye on the [printtime_left] that I have added into the title bar display, for the moment.
[image]

OllisGit, 2022-01-12:
duplicated, see #249

closed 2022-01-12 by OllisGit