OllisGit/OctoPrint-DisplayLayerProgress #246: Height / Total Height issue

nilshomeier, 2021-10-31:
Hi!

I've got an issue with the height reporting of the plugin. It seems as if its showing the current layer divided by 10 as current height and no total height at all.
I'm using Octoprint 1.7.0 on octopi 0.18.0 and plugin version 1.27.1. Slicer is Superslicer 2.3.57.

Thanks!

[image]
[FlowTest_0.20mm NORMAL - PLA_extrudr NX2_10m (1).gcode.zip](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/7448328/FlowTest_0.20mm.NORMAL.-.PLA_extrudr.NX2_10m.1.gcode.zip)

nilshomeier, 2021-10-31:
I've also enabled debug. Maybe it helps?

[plugin_DisplayLayerProgress_events.log](https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/files/7448367/plugin_DisplayLayerProgress_events.log)

OllisGit, 2021-11-15:
Hi @nilshomeier, 
the max height of the print is analysed by OctoPrint and stored in a file called ```.metadata.json``` located in the upload folder.
Should looks like this:
```
    "FlowTest_0.20mm_NORMAL_-_PLA_extrudr_NX2_10m_(1).gcode": {
        "hash": "8ecf4349ea8f4e5ad19e4574759ad45cd5b93463",
        "display": "FlowTest_0.20mm NORMAL - PLA_extrudr NX2_10m (1).gcode",
        "analysis": {
            "printingArea": {
                "maxX": 168.909,
                "maxY": 164.849,
                "maxZ": 2.2,
                "minX": 10.0,
                "minY": 4.0,
                "minZ": 0.2
            },
            "dimensions": {
                "depth": 160.849,
                "height": 2.0,
                "width": 158.909
            },
            "estimatedPrintTime": 432.73752605944816,
            "filament": {
                "tool0": {
                    "length": 510.80624000001194,
                    "volume": 1.2286329909156446
                }
            }
        },
```
If the analyse is missing, then please check the "Feature" settings 
[image]

Then it should looks like this
[image]

nilshomeier, 2021-12-04:
Thanks! Apparently it has to be set to "Always". Mine was set to "Only when idle". It now works as it should!

closed 2021-12-04 by nilshomeier

luizbgomide, 2022-01-07:
@OllisGit I'm also having this issue of DisplayLayerProgress reporting "-mm" as total height, but I'm not sure what is causing it. 

The model has the information required:
[image]

I've set Analyse Gcode to always. But still it says `analysisPending: false`??
[image]

Any thoughts?

OllisGit, 2022-01-12:
hmm..I searched the OP-Sourcecode for "analysisPending" value, but without success.
Which version of OP do you use?

Btw. please raise a new issue (with attached gcode and octoprint.log), because this one is closed.

luizbgomide, 2022-01-12:
@OllisGit nevermind. I managed to "fix" the issue. I removed PrintTimeGenius and it all worked.

So I cleaned all its data, reinstalled print time genius and now everything is working just fine... Somehow PrintTimeGenius was hijacking the analysis and never actually returning.

OllisGit, 2022-01-12:
Good to know, thx. Have a nice day!