OllisGit/OctoPrint-DisplayLayerProgress #249: Estimated End Time not respecting time zone

Sebastian-Schneider, 2022-01-10:
ETA does not respect the users time zone:
[image]

It is always 1h off from the actual ETA. I am located in Germany with at CET (+1h), the time displayed is the ETA at GMT time.

Im running a standard OctoPi installation (did not change anything in the system, only installed some plugins). Client is Ubuntu with up to date Firefox, timezone correctly configured.

bkbilly, 2022-01-12:
I've got the same problem. I am in Greece at CET (+2h) and the estimated hour is 2 hours before the actual time.

koulis-probotek, 2022-01-12:
I think that I found the issue with the time for my instance of octoprint.
I am using a docker container to run Octoprint and the timezones are set to UTC.
The solution that I am going to try is to link the files that are responsible for the timezone from my system with docker by changing my docker-compose file like so:
```yaml
    volumes:
      - "/etc/localtime:/etc/localtime:ro"
      - "/etc/timezone:/etc/timezone:ro"
```
I will let you know if it works and I hope @Sebastian-Schneider that this works for you too.

bkbilly, 2022-01-12:
I think that I found the issue with the time for my instance of octoprint.
I am using a docker container to run Octoprint and the timezones are set to UTC.
The solution that I am going to try is to link the files that are responsible for the timezone from my system with docker by changing my docker-compose file like so:
```yaml
    volumes:
      - "/etc/localtime:/etc/localtime:ro"
      - "/etc/timezone:/etc/timezone:ro"
```
I will let you know if it works and I hope @Sebastian-Schneider that this works for you too.

Sebastian-Schneider, 2022-01-12:
I'm running Octoprint within an unmodified OctoPi instance on a Raspberry Pi. I updated timezone by running `sudo raspi-config` under "Localisation > Timezone", the default "London" was there before. After the change, the two mentioned files by @bkbilly were updated and this fixed the issue for me too.

Thank you @bkbilly for the hint!

@OllisGit This is resolved for me, but you could consider to calculate the ETA in the frontend too, as other plugins do. Anyways the official OctoPi setup documentation mentions the necessity to set the correct time zone as "optional". I missed that.

OllisGit, 2022-01-12:
fyi: The plugins use different mechanism for calculating the time(as already mentioned from @Sebastian-Schneider):

- "DLP" time is calculated on serverside in python. So the server environment is used. This is necessary because other plugins and clients reads this information 
- "Dashboard" plugin calculates the time in your local browser via JavaScript

An other good solution is to bind your RasPi to a TimeServer (NTP or "FritzBox" also provides an NTP-Server)

closed 2022-01-24 by OllisGit