OllisGit/OctoPrint-DisplayLayerProgress #252: Feature-Request: Use M118 instead of M117 commands

OllisGit, 2022-02-03:
Details see https://community.octoprint.org/t/how-to-send-m118-a1-p0-action-notification-time-left-hms-e-g-02h04m06s/41553

OllisGit, 2022-02-06:
A dev version that just replace all M117 comands with M118 is out: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.29.0.dev1

Pascal-m, 2022-02-07:
Hy @:OllisGit,
Thank you for the version of Dev
I started tested and I detected a small dysfunction in the remaining time.
TFT is waiting for a non-truncated time format :

`M118 A1 P0 action:notification Time Left hms (e.g. 00h16m06s)`

Unfortunately when I use the  [printtime_left] statement the time is cut example for a time remain 20 minutes 13second I receive 20m13s while I would have that I receive **00h20m13s** 

Is there a way to keep hours the minutes and seconds? 

Thk :)

[image]

OllisGit, 2022-03-23:
Hi @Pascal-m,
for seconds there is already a flag (in your case you need to disable it), but not for the leading minutes/hours.
[image]
 
I need to implement this feature

MehStrongBadMeh, 2022-04-04:
> I need to implement this feature...do you need two digits or is one enough? I prefer one digit, like this `0d0h0m31s`

I just tested this out by manually sending M118 commands via Octoprint Terminal. Sending the command `M118 A1 P0 action:notification Time Left 1h1m10s` and `M118 A1 P0 action:notification Time Left 0h1m10s` both properly updated the time remaining. So it appears there is no need for there to be two digits. Same goes for using M117, as I don't have the UART issue with my setup, so only the lack of hours when less than one hour remaining is an issue.

Pascal-m, 2022-04-04:
hello sorry for the delay
In the specification it takes 2 digits on the hour minute and second

example: 
01h02m06s 
Or
00h09m04
Thx

Pascal-m, 2022-06-05:
> 

Hello OllosGit,
good news, in the latest version of the TFT firmware, it accepts one digit now   :)
Do you think you can make the change?
Thk 
Pascal

> Hi @Pascal-m, for seconds there is already a flag (in your case you need to disable it), but not for the leading minutes/hours. [image]
> 
> I need to implement this feature...do you need two digits or is one enough? I prefer one digit, like this `0d0h0m31s`

wd5gnr, 2022-10-15:
Just to let you know, I am using 1.29.0.dev1 with the same use case and for layers and progress, it works well. I haven't dug into the time yet since there are only two notifications.

racSF, 2022-10-19:
> Just to let you know, I am using 1.29.0.dev1 with the same use case and for layers and progress, it works well. I haven't dug into the time yet since there are only two notifications.

True - dataprogress is correctly set, when using

A1 P0 action:notification Data Left [current_layer]/[total_layers] 
as message pattern.

I can also confirm, that the issue with the missing leading hours still exists, would it be possible to implement it as option like the one mentioned by pascal-m ? For this I would use:

A1 P0 action:notification Time Left [printtime_left]

racSF, 2022-10-20:
I'll try to fit my needs by altering the file: "/home/pi/OctoPrint/venv/lib/python3.9/site-packages/octoprint_DisplayLayerProgress/stringUtils.py"

And change the secondsToText function to:

def secondsToText(secs, hideSeconds=False):
    result = ""
    days = secs // 86400
    hours = (secs - days * 86400) // 3600
    minutes = (secs - days * 86400 - hours * 3600) // 60
    seconds = secs - days * 86400 - hours * 3600 - minutes * 60
    if (hours > 0):
        if (hideSeconds == True):
            result = "{}h".format(hours) + "{}m".format(minutes)
        else:
            result = "{}h".format(hours) + "{}m".format(minutes) + "{}s".format(seconds)
    elif (minutes > 0):
        if (hideSeconds == True):
            result = "{}m".format(minutes)
        else:
            **result = "{}h".format(hours) + "{}m".format(minutes) + "{}s".format(seconds)**
    elif (seconds >= 0):
            **result = "{}h".format(hours) + "{}m".format(minutes) + "{}s".format(seconds)**

    # result = ("{}d".format(days) if days else "") + \
    #          ("{}h".format(hours) if hours else "") + \
    #          ("{}m".format(minutes) if not days and minutes else "") + \
    #          ("{}s".format(seconds) if not days and not hours and seconds else "0s")
    return result

This should only help if you use the sidewinder - I'll test it maybe tommorrow ;)

heinrichWeichert, 2023-02-20:
I'm also interested in this, could this be made possible with a variable like [printtime_left_incl_hours] ?

raihei, 2023-02-20:
That would be great, I am also interested. And maybe a checkbox "Use M118 instead of M117"?

lethuer, 2024-03-28:
I also want to use M118 instead of M117 for my tft on an artillery genius.
Is there any action to get an "official" update to provide this possibility in the plugin ?

lethuer, 2024-03-29:
Hi there !

I'm using an Artillery Genius (**Not** Pro).
I tested this test version for M118 provided by @OllisGit: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/tag/1.29.0.dev1
together with this small modification provided by @racSF: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/252#issuecomment-1289512786
and created a zip File with that which can be installed directly.

I have configured this:
For print progress I used layer progress information: P0 A1 action:notification Data Left [progress]/100
Another possibility would be to use layer information: P0 A1 action:notification Data Left [current_layer]/[total_layers]
[image]

That's working for me AFTER uploading a file again (which includes M118 messages instead M117), so I would really like to get a stable version !
Otherwise changes would be gone with next update, which is provided:
[image]

Additionally I would like it would be possible to send **three** M118 messages to the printer:
```
M118 P0 A1 action:notification Time Left XXhYYmZZs
-> Updates Time Left Field
M118 P0 A1 action:notification Layer Left XX/YY
-> Updates Layer Left Field
M118 P0 A1 action:notification Data Left XX/YY
-> Updates Percentage Field + Progress Bar
```
But currently there are only 2 field available.
For me it wouldn't be neccessary to toggle between messeges, they could be send all at the same time.

lethuer, 2024-03-29:
> One small downside:
> Currently I additionally have installed an SSD1306 display together with this plugin:
> https://plugins.octoprint.org/plugins/ssd1306_oled_display/
> With the modified "DisplayLayerProgress" plugin mentioned above on the display now "INIDICATOR-Layer1" is displayed.
> I think the reason is that the plugin also uses M117 messages which are now completely replaced by M118.

With solution discussed in OctoPrint forum it is possible to stay with original "DisplayLayerProgress" plugin and use a modified "M73 Progress" plugin.
https://community.octoprint.org/t/tft-percentage-progress-not-updatet-m73/57420/27?u=lethuer
This makes my previous mentioned downsides obsolete:
1. All M118 messages are sent at the same time
2. Other plugins which also requires M117 messages aren't affected

lethuer, 2024-03-30:
For adding layer information I've found this possibility to use in the slicer: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/wiki/How-does-the-plugin-works#cura
Unfortunately this modified script "InsertAtLayerChangeEnhancement" don't use the information about [LAYER_COUNT] yet.
So this has to be updated to get a format like needed:
`M118 P0 A1 action:notification Layer Left [layer_num]/[LAYER_COUNT]`