OllisGit/OctoPrint-DisplayLayerProgress #254: undervoltage/alert-icon on wrong side of navbar_plugin_pi_support

newhinton, 2022-02-19:
The issue-icon that only shows up when the pi has detected an issue with itself is on the left side of the Layer Progress. It should be on the right side with all the other icons. Is it possible to change that?

[image]

newhinton, 2022-03-22:
Not stale

newhinton, 2022-04-22:
This is still not stale. Those stale-bots are really annoying, just because this ticket has no interaction because noone is working on it for a month, does not mean it is fixed or should be closed.

Now i have to create a completely useless comment just to keep it alive.

cp2004, 2022-05-07:
@newhinton - you can change the order of the components yourself in OctoPrint's config.yaml, under the `appearance.components.navbar`

https://docs.octoprint.org/en/master/configuration/config_yaml.html#appearance

Adding just the `DisplayLayerProgress` entry should be enough to bring it to the beginning, as described in the docs above.

I am not sure if there is a way for plugins to suggest the order, this would be hard to do without conflicts I guess.

closed 2022-06-17 by github-actions[bot]

OllisGit, 2022-07-09:
@newhinton or you can use https://plugins.octoprint.org/plugins/dragon_order/