OllisGit/OctoPrint-DisplayLayerProgress #257: Current Height

CmdrCody51, 2022-03-13:
It just totally ignores relative motion in the - direction.
Quick test Control Tab
Jog  Z up 10 (displays Height: 10)
Jog  Z up 10 (displays Height: 20)
Jog  Z down 10 (displays Height: 20)
Jog  Z up 100 (displays Height: 120)
Jog  Z down 100 (displays Height: 120)
You can go up and down all day until your Height is 5K.

OllisGit, 2022-07-11:
Hi @CmdrCody51,
thanks for reporting the issue.

I fixed this and it is now available in version 1.29.0rc1. 

If you are using OP 1.8.x you can switch between the release-channels via "Software Update" menu.
Or you can reinstall the version via plugin manager: https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/releases/download/1.29.0rc1/master.zip

Please test and give feedback.

Thx, in advance
Olli

CmdrCody51, 2022-07-12:
Oh Oh - I am all a quiver.....IT WORKS!