OllisGit/OctoPrint-DisplayLayerProgress #263: No more compatible with Octoprint 1.8.

FonceSlimane, 2022-06-03:


FonceSlimane, 2022-06-03:
Hi, this usefull plugin hangs and crashes Octoprint new version 1.8.0 and 1.8.1 on my Raspberry Pi3A
Octoprint lags and becomes unreachable now
Power on reset can solve the problem only temporally for a few seconds.
Safe mode of Octoprint works fine.
After troubleshooting, deactivating the plugin DisplayLayerProgress solved the problem.

elaurijssens, 2022-06-07:
Same here, DisplayLayerProgress brought OctoPrint to a grinding halt, with one of the cores on a Raspberry Pi4 maxing out at 100%. Removing the plugin solved the issue.

OllisGit, 2022-07-09:
@FonceSlimane @elaurijssens 
Please attach the ```octoprint.log``` maybe I can detect an issue.
Btw. I installed a fresh OP1.8.1 on my dev-system and could not see any issue.