OllisGit/OctoPrint-DisplayLayerProgress #264: No messages on Ender 3 with Creality touchscreen upgrade

find-a-good-nickname, 2022-06-06:
I have a Creality Ender 3 (not Pro) with v4.2.2 stock board and a Pi 3B+ with Octoprint and DisplaLayerProgress plugin. Everything went flawlessly on the stock display (with that crazy knob) with the _"[printtime_left] left. L=[current_layer]/[total_layers]  [current_print_filename] [progress]%"_ printer display message.
I bought a Creality Touchscreen upgrade (https://www.creality3dofficial.com/products/upgraded-touch-panel-screen-display-for-ender-3-pro-ender-3-v2), but there's no visible printing messages at all on the screen, when I print from OctoPrint. The configuration of DisplayLayerProgress is the same as before, the started printing runs OK, but I'd like to see print time left etc.

OllisGit, 2022-07-09:
Can you send M117 (e.g. enter ```M117 Hello```) messages via OP Terminal to the display?

Some firmwares/displays ignore M177 commands, see https://github.com/OllisGit/OctoPrint-DisplayLayerProgress/issues/18

find-a-good-nickname, 2022-07-10:
I tried to wrote "M117 Hello" to the terminal page of Octoprint, the little pale blue "Printer Display" shows the hello on the left lower corner, but absolutely nothing can be seen on Ender 3 touchscreen.
I wrote also "M177 Hello" to the terminal, but -as you said- the answer is 
>>Recv: echo:Unknown command: "M177 HELLO"<<

OllisGit, 2022-07-10:
Hi @find-a-good-nickname ,

you sent ```M177``` and not ```M117```.

The browser printer display shows only M117 commands which are already sent from OP to the printer, but it doesn't care about the response of the printers fw.

Maybe there is a an other gcode that could be used to display messages in your display.

Did you already ask the "Creality-Community": https://forums.creality3dofficial.com/community/ ?

find-a-good-nickname, 2022-07-10:
Misunderstood.
I tried M117 AND then M177 too, but after M117 there was nothing on the printer touchscreen (but the browser's mini display said "hello"). And after M177 there was also nothing on the printer screen, but there was that error message on the browser terminal.

I only asked on the plugin page, because I thought that this problem is plugin-related. :)